hildogjr/KiCost #10: Allow custom pricing for non-distributor parts

xesscorp, 2015-09-30:
Import of custom part pricing from some source (CSV, command-line, etc.)   This one is not as high because at some point some of this custom stuff is not work the dev time to setup or maintain.  Part is influenced by running the script a lot because I'm trying to fix things -- now that I'm fairly fixed - I think it's usable to start manually editing the spreadsheet I have rather than generating more.   However, I'll throw this out there.  Many times I have to spec a clock part which is a custom SKU with parameters not available from distributors.  I therefore get the clocks direct from the manufacturer and these will never cross-ref. pricing at any distributor.   The thought would be some mechanism for defining the cost of these parts and get these sucked into KiCost when a run is made.   My process right now is going to be to fill in the price by hand based off of quotes I have in email.

stevenj, 2015-11-24:
I think the easiest way to deal with this is just add a custom field to the part "Price#", which is the manual price and another "PriceSource#" which is a link to the source of the price, this is optional. Or something like that.  That way if you want those in your BOM, you add the fields to components that can't otherwise be found.  Then there is no other source that needs maintaining, just the schematic.

karrika, 2015-11-25:
I already have a field "Price" for the components that I buy from odd vendors. I really like this "Price#" and "PriceSource#" idea. In this way I can also create dummy components with no footprint in the schematics eg. the PCB itself and it will pop in the final spreadsheet as a custom part.

xesscorp, 2015-12-30:
I added this feature in version 0.1.13. There are two new fields: kicost:pricing and kicost:link for storing the quantity pricing and the documentation link for a part. You can also add a distributor name in the middle of the field label to indicate the source for the part, such as kicost:my_weird_dist:pricing.

closed 2015-12-30 by xesscorp