hildogjr/KiCost #4: Display "extra" component data

xesscorp, 2015-09-30:
For "extra data" information, xlsx could display some distributor site return data (value, tolerance, footprint/package, etc) which doesn't necessarily care about a match to the component data.  Although maybe cares about consistency or a fuzzy match between distributors (mouser, digikey, Newark)

Which, if doing that, begs for a template file (text file) to setup which fields to display as "extra data".   Maybe instead of relying on fuzzy matching to have the setup file also prioritize which site "wins" if information exists.   Example: setup file wants two "extra data" fields and perhaps denotes them as:

$extra1=mouser,"tolerance",digikey,"tolerances",Newark,"tol:";$extra2=digikey,"package"

Example shows two extra fields.  First field will fill in with a match on part html line with "tolerance" match.   If no match, then try Digikey a match for "tolerances" (showing maybe different search strings for different distributors), if no match then try Newark a line with "tol:".   The 2nd extra field only matches if digikey returns a field with "package".   How the script decides what the value is to return (presumably data to the right of the search string) - how much to return (and when to terminate) may not be worth deciding so there may need to be a third field which tells the script how to determine.   $extra1=mouser,"tolerance","regex? Something else",digikey,"tolerances","dk value parse helper", Newark."tol:."newark parser helper"

Not settled on the parser helper field - but that it could help the script abstract how to determine the value return format.

hildogjr, 2017-12-28:
Related #78 (Add datasheet link), #91 (Show part image)
It will be used in #2 (Part value auditing)

hildogjr, 2018-06-07:
Missing implement on others distributors, use my code Digikey as example/model.

hildogjr, 2018-10-07:
USinf #295, @xesscorp , would be interesting to add the life cycle (I intend to use some highlight in the spreadsheet if not recommended to new designs)  and datasheet to complete that parts that doesn't have one.

hildogjr, 2018-10-17:
Actually use the life cycle and datasheet to complete and highlight the `manf#` cell code.
New information could be added in the future if available by the web distributors (now APIs) and a cross-checking could be implemented.

I will leave this opened for now until I study if is possible this cross checking.

hildogjr, 2018-11-06:
If this word parse https://kitspace.github.io/electro-grammar/ should be possible to validate.

commit 633f3593 (2018-03-24): Merge pull request #220 from hildogjr/master

Fix #4 (prototype on digikey)
commit d603171c (2018-03-24): Fix #4 (prototype on digikey)