hildogjr/KiCost #5: Part data entry via spreadsheet

xesscorp, 2015-09-30:
Allow entry of BOM type data (manufacturer/manufacturer part number) into a spreadsheet (CSV) and allow import of this into the kicad project files so I can fill in manf and manf# using a spreadsheet rather than Kicad.

Arunmaran, 2016-02-18:
Hi xesscorp,
How to import the spreadsheet in kicad for manf# field.

xesscorp, 2016-02-18:
The separate [KiField](https://kifield.readthedocs.org/en/latest/readme.html) tool now provides the ability to enter field data into a spreadsheet and insert it into a schematic or library.

closed 2016-02-18 by xesscorp

Arunmaran, 2016-02-18:
Hi xesscorp
I got it. Thank u .

Apart from kicost and kifield any other features are there for kicad.

xesscorp, 2016-02-18:
Currently, the only other utility is [KiPart](https://kipart.readthedocs.org/en/latest/) which helps make schematic libraries from spreadsheets containing their part pin information.