hildogjr/KiCost #6: Ignore certain parts using their reference prefix as a filter

xesscorp, 2015-09-30:
The other items would be to allow a command-line option to ignore certain refdes alpha-parts.   Since different people may have different conventions, I would have a line like:

c:\python27\scripts\kicost -i my schematic.xml -g tp,dh

Where after -g ("i" was already taken - g for iGnore) is a comma separated list of "whole word" alpha portions of the reference designator to ignore and case insensitive.   TP* for me are testpoints.  DH* for me are drill holes.  If I were to do a -g t,dh in this example, then the "-g t" would not ignore "TP" as the "t" has to match the entire alpha part before the number part.  Exact case-insensitive match.

I just took all of my TP and DH parts and assigned INSTALL=N which tells my script not to assign manf/manf# fields to these parts.  I was already doing this since my CSV file which is used as a lookup table to inject manf/manf# into the schematic did not have TP or DH refdes(es) in the CSV which also served to keep manf/manf# from being injected.   But the idea that occurred to me is that if there was a switch to ignore blank manf/manf# so the spreadsheet just wouldn't list these - that would work as well.   It'd be better not to have to filter these out by doing extra work on the front-end (better to use the -g switch which can be scripted and requires no extra effort) - but having the option to remove blank manf/manf# parts from the list would be good. Maybe -b (- for ignore b for blank).   I'd normally run without this switch at first to see if there's anything missing a manf/manf# pair that I thought should have one.   But then -b for subsequent runs.

cb88, 2015-12-16:
Also, I sometimes have DNI in the manf# field... for Do Not Install. This is usually for a part that isn't populated and/or doesn't have a component selected yet but does have a footprint on the board for future use.

NF6X, 2016-03-19:
I found that when I had DNI in the manf# field of a part, the script found some wacky part with a high cost at one or more of the vendors and this inflated the cost estimate by several hundred dollars. I removed DNI from the manf# field of the DNI part, leaving it just in the value field.

hildogjr, 2017-12-28:
Using the --ignore_fields

closed 2017-12-28 by hildogjr