hildogjr/KiCost #13: Errors/not running files

zegarac, 2015-10-28:
Hello,
I have a problem not runnig some bom's;
WIN10, with kicost installation and drag&drop batch file (i just drag xml file and drop over KiCost.bat to do the job)
KiCost.bat file content:
""""""""""""""""""""""""""""""""
@echo off
set path=C:\python27
set SOURCE=%1
cd C:\Python27\Lib\site-packages\
python kicost -v
@echo File to process: %SOURCE%
python kicost -i %SOURCE%
pause
"""""""""""""""""""""""""""""""""""""""""""""""""
Certain xml bom files are not running, throwing out errors, other BOM XML files are running fine (smaller stuff) paste of the error:
""""""""""""""""""""""""""""""
KiCost 0.1.10
File to process: "C:\Users\blanked\Desktop\Handheld board.xml"
Traceback (most recent call last):
  File "C:\python27\lib\runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Python27\Lib\site-packages\kicost\__main__.py", line 84, in <module>
    main()
  File "C:\Python27\Lib\site-packages\kicost\__main__.py", line 80, in main
    kicost(in_file=args.input, out_filename=args.output, debug_level=args.debug)
  File "kicost\kicost.py", line 93, in kicost
    parts = get_part_groups(in_file)
  File "kicost\kicost.py", line 133, in get_part_groups
    for p in root.find('libparts').find_all('libpart'):
AttributeError: 'NoneType' object has no attribute 'find_all'
Press any key to continue . . .
"""""""""""""""""""""""""""""""""""""""""""""""""""""

It is fairly large BOM with 370 components, i did add "manf#" in there as a field (it is too big to paste it here) plus does not take xml when dragging&dropping here...
The app is running some other XML bom's which is weird...

example from xml:
"""""""""""""""""""""""""""""""""
 <comp ref="U31">
      <value>OPA2228</value>
      <footprint>w_smd_dil:so-8</footprint>
      <datasheet>OPA2228UA/2K5</datasheet>
      <fields>
        <field name="manf#">OPA2228UA/2K5</field>
      </fields>
      <libsource lib="Handheld board-cache" part="OPA2228"/>
      <sheetpath names="/ref_detector/" tstamps="/55F41B07/"/>
      <tstamp>55E61ADA</tstamp>
    </comp>
""""""""""""""""""""""""""""""""""

xesscorp, 2015-10-28:
Thanks for trying KiCost!

Try running KiCost in a command window and use the option "-d 2". This will provide a bit of debugging info as KiCost processes the XML file. Maybe that will pinpoint where the problem is occurring. Then send the XML file and the error trace and I'll look at it.

xesscorp, 2015-12-03:
No response. I'll assume this issue is closed.

closed 2015-12-03 by xesscorp