hildogjr/KiCost #14: Support for RScomponents

karrika, 2015-11-14:
I would like to add support for RScomponents.

kasbah, 2015-11-14:
Just FYI, if you output in a format compatible with my extension: [monostable/1clickBOM](https://github.com/monostable/1clickBOM), you could support all the retailers it supports which includes RS.

FlYnN91, 2016-02-02:
+1 for this

glcerone, 2016-12-14:
From version 0.1.31 Kicost supports also Farnel and RS Components.

nickoe, 2017-12-24:
Why is this bug not marked as closed?

hildogjr, 2017-12-24:
I don't know. But there are a lot of others in same status (already fixed / implemented but opened yet).
I think, really active (in discussion or implementation) we have 10 or less (and 40 opened / the author of the issue forgotten to close).

hildogjr, 2017-12-24:
Can @xesscorp (owner of the directory) close issues or just the author?

closed 2017-12-27 by hildogjr