hildogjr/KiCost #15: All Prices 0 in spreadsheet

anderwm, 2015-11-16:
Hello, I like your project.  However, I am getting all pricing information coming back as 0.00.  All the links are correct and it seems to scrape correctly.  It just doesn't seem to put the prices back into the worksheet.  Any ideas what I could be doing wrong?

xesscorp, 2015-11-19:
I just tested kicost and the spreadsheet looks ok. I'm using LibreOffice 5.0.2.2 but I've also used versions previous to this. What OS and LibreOffice version are you using. Can you send the .xml file and the .xlsx file?

anderwm, 2015-11-19:
[PowerModuleBOM.xlsx](https://github.com/xesscorp/KiCost/files/39031/PowerModuleBOM.xlsx)

[PowerModule.txt](https://github.com/xesscorp/KiCost/files/39032/PowerModule.txt)

I had to change the extension of the xlm file to get it to upload, just change it from .txt back to .xlm.  I am also using 5.0.2.2 so I'm not sure the issue, I didn't get so far as to look at the actual formulas in the cells to troubleshoot.

What you have done is really great, I've done screen scraping before and it isn't easy to keep it working if you don't do a good job initially(which you obviously did).  Thanks for sharing this.

xesscorp, 2015-11-19:
I opened your spreadsheet and everything displays fine for me (as far as I can tell).

I think this is a problem caused by spreadsheet recalculation. I recall having a similar problem myself during development. Try pressing Ctrl+Shift+F9. This will do a "hard recalculation". Then see if the pricing information appears.

I actually have a hard recalculation button that I placed on my Calc toolbar to make this easier.

anderwm, 2015-11-19:
Your right this solved the problem.  It is something to do with BoardQty variable because if I change a cell with BoardQty that line begins to work.  Thanks.

closed 2015-12-02 by xesscorp