hildogjr/KiCost #17: Enhancement: Supporting Generic line items

stevenj, 2015-11-24:
Things like resistors and caps are quite generic.  It would be nice if there was a way for kicost to lookup these types of components based on their parameters (type, value, package).  And then pick the most cost effective one for that job.  Not sure the best way to achieve that, but it would be nice.

If the suppliers cant be directly queried based on the parameters than another way would be to have a generic part database, CSV for example, which maps the component parameters to known parts, but then someone would have to maintain that.  However, it could be locally maintained, so that once you have verified that the Yageo 0.1uf Cap is just as good as the Murata 0.1uf Cap you add that to your own local list of Generics.

Just an idea.

hildogjr, 2017-08-29:
I would link to link #77 with this request. I think if we standardize the code, we may implement a module / function that getting the value / reference (capacitor, resistor), if the manf# (and other) is empty, parse one string or use some definitions / dictionaries to pass the string to look for in the distributors to search capacitors / resistors.

hildogjr, 2017-08-30:
I did some manual test in the web sites and with `100nF ceramic 0805` (value material and package/footprint) I could found the capacitor that I want.
I think this is the way: one python module/file specific do deal with (using generic type of material names) some definitions and with this three data parse one `manf#` to the rest of the `KiCost` script.

xesscorp, 2017-09-04:
I don't know if this is an appropriate function to place in KiCost. KiCost generates part cost spreadsheets. The requested function is more like doing part selection. I could see a separate tool that processes an XML file and inserts part numbers for certain types of parts like resistors/capacitors. Then the augmented XML file could be passed to KiCost. That keeps the complexity of part selection separate from the complexity of costing.

hildogjr, 2017-09-04:
Maybe as separated module active by `--option`in the call, that is default off?

hildogjr, 2018-01-02:
@stevenj , I thinking about this implementation as a module that get informations in fileds of KiCad as 'value', 'manf', even footprint.
In some conceptual test I look for in digikey for " 390R 5% 1/4W axial" and I got almost the "CF14JT390RCT-ND" that was one that I had select in the past to this specifications.
I am thinking in an additional module to KiCost that we could specify the default characteristics (such as "tolerance", ...) but, I need help of the users to test each distributor web site and see if this kind of search string return always useful results (so KiCost got the most cheaper).
And of courser, first the implementation said by me in #4.

hildogjr, 2018-01-13:
@xesscorp , the distributors sub modules have not yet my full understanding. How exacting these is working, just with a full exactly matching or analyzing the the answer of a match.
Explaining, for manf#=UA78M05CDCY, digikey page in my spreadsheet is https://www.digikey.com/product-detail/en/texas-instruments/UA78M05CDCY/296-27138-5-ND/1910303, but search on the site appear a list https://www.digikey.com/products/en?keywords=UA78M05CDCY%20.
The part is got by the exactly match in the results? (Understanding the possibility of search of `value + footprint + voltage_max + tolerance + ...` and got some results).

hildogjr, 2018-01-22:
In each distributor submodule there are the code `difflib.get_close_matches` to the case that a search result a table of results and not just a single match.
This could be modified (by a `get_part_html_tree()` input parameter) to get the most shipper one and not the close match (close match still the default). So, using this and the informations that I told in the last comment (this I can extract by the footprint name and some definitions given by the user, I have some idea), I think KiCost will be able to select capacitor and resistors.
Maybe, the start point, is using the issues #146 and #161 and create this modification.

mobilinkd, 2018-11-17:
I'd like to suggest a potential half-way step to specifying generic components.  Allow users to specify alternate part numbers in the manf# field, with parts separate by a pipe symbol (logical OR operation).

For example:

    manf#: SKSCLBE010|EVQ-P7C01P

This would allow kicost to price either of these two tactile switches.

Another example is this case:

    manf#: FC-135 32.7680KA-A0|FC-135 32.7680KA-A3|FC-135 32.7680KA-A5

These are the same part, distributed on different sized reels.  I use other components that are distributed either on trays or reels, each with distinct part numbers.

I see that #236 was closed as "wont fix" without any comment as to why.  To me it seems that #236 is less involved and solves a more practical problem (at least for me) than allowing one to specify generic components.  `get_close_matches()` may solve the second issue, but I done see how it will solve the first.

I have a number of BOM items that have qualified alternates (dual MOSFETs, tantalum caps, polyfuses, EEPROMs, tactile switches, etc).  Other than the tantalums, I don't know if anyone would consider any of the others to be a *generic component*.  I would not want kicost to price them that way.

hildogjr, 2018-11-18:
@mobilinkd , I moved this discussion into #236 to not spam or mix the subjects.

craigarno, 2020-08-14:
Reading through this thread has given me a new perspective on component purchase and I can see some merits to providing KiCost with an exact manufacturer selection.  I have been in design situations where this is exactly the behavior desired and needed.  So let me try to give the other perspective.  This is long, I'm sorry, but it's important.

As a developer either sole, or in a group of developers, there are times when in-house parts won't satisfy new design part needs.  The list of "new" parts could number in the 50-100 range.  It takes a lot of time to individually search Digikey, Newark, Farnell, Mouser and other supported **vendors** for these new parts, and if two or more of these do carry the needed single manufacturer's part, they don't usually have the same prices.  Cost -by- **Manufacturer** can and often is part of the new part selection process, or old part selection review.  Let me provide a simple to follow example:

New part "**555 Timer**" (old tech but new to company).  In the schematic this is listed as **NE555** which is functionally compatible with **TLC555** and a host of other part numbers from different **manufacturers**.  From a design point of view I'd like a "_loose_" search for part matches.  for instance if I go to Digikey and search for "**555 Timer**" I get matches for **Manufacturers**; _TI, National, Renesas Electronics, STMicroelectronics, Microchip Technology, Diodes Incorporated, ON Semiconductor, Maxim Integrated, Nexperia, Rochester Electronics, Advanced Linear Devices, Rohm Semiconductor, NXP, NJR_.  This is great, it's what I want, and prices on Digikey vary considerably between manufacturers for functionally equivalent parts.

This price variance is why part selection must consider **manufacturer selection** cost in addition to **vendor cost**.  If I make a spreadsheet and label columns _Digikey, Mouser, Newark, etc._ then for rows I put _TI, National, Renesas Electronics, STMicroelectronics, Microchip Technology, Diodes Incorporated, ON Semiconductor, Maxim Integrated, Nexperia, Rochester Electronics, Advanced Linear Devices, Rohm Semiconductor, NXP, NJR_.  It is easy to see that I need to investigate **5 vendors** times **14 manufacturers** or **70 different prices** to provide the data needed for a **manufacturer+vendor** combined cost decision.  Now if you have 100 new parts which also need this process, that is **7000 part-manufacturer, vendor prices** which must be analyzed for one "new" design.

At manufacturer level the Design Engineer must be involved in selecting a manufacturer because different manufacturers use different process technologies which can result is different setup/hold and propagation delay timing which will determine if a circuit design will work or not, even if the part is functionally the same and pin compatible.  comment: I've had this happen with memory chip manufactures.

Other factors which prevent selecting "bottom dollar" parts fit in some of these categories;
- delivery time vs quantity in a JIT type environment
- Second and Third Vendor/Manufactuer sourcing company procurement requirements
- temperature range; commercial vs automotive vs military
- MTBF

I don't expect KiCost to help with all of these decisions, but would hope KiCost can help with manufacturer selection, especially with simpler part selection.  i.e. a column checkbox for "exact manufacturer match" which if unchecked will include other possible manufactures (loose search), just like Digi-key and Mouser do with online component searches.

Summarizing, not including Manufacture in the Vendor selection process is like choosing product purchase price without considering $25-$100 shipping cost.  You can make a bad decision by only looking at half the data.  Which to use (loose vs exact) will depend on if part searching is done by a procurement specialist, or a design engineer.  Design Engineers can do initial selection if the tools make it easier with automation.  Otherwise the design engineer is left doing initial Manufacturer selection manually which is quite tedious.  Having KiCost do this initial "loose" part-manufacturer search with clickable datasheet links for detail component specifications is quite useful.  Right now I see KiCost as solving the procurement problem, not the manufacturer problem.  I think it could do both without too much effort depending on who is running the tool.

Thank you for reading this far.

noafterglow, 2026-02-07:
My companies went through a series of this exact kind of thinking over the course of 20 years of product development, through several electronics supply chain boom and bust cycles, in flash memory, EOL of various discrete and passives, etc.  One included a shortage of 10K resistors, (not making this up).  This post is long on explanation, but the end result is that the engineers need to greatly LIMIT the number of passives line items used, and alternates are a VERY good idea.  In addition "standard" caps at the higher spec range of  voltage, tolerance, and material of each size are the ONLY caps to spec and use (more on this in a video soon).

We settled on a hybrid part field/name strategy, including a giant pile of fields: 
-- putting multiple MANF# in the part# field, and including the actual MANF name.  We actually used separate fields (MAN1, PN1, MAN2, PN2), as this allowed people to know the brand before picking up the phone.
-- Labeling EVERY part with an OPEN_BRAND field with choices of: (yes, no, maybe (check with us), single_source)
-- Required, dedicated, spec fields for passives, beyond size, and footprint.  These included, for R/C:
    -- R:  Value, Tolerance, Temp-co, Composition (Thick film, etc), Wattage.
    -- C:  Value, Tolerance, Operating Temperature, Composition (X7R, Aluminum Electrolytic, etc), size (not just footprint), Voltage.
-- An OPEN_BRAND_MATCH field which specified which parameters must match or be better.  ex for a 0603 resistor: (Resistance, Tolerance, Footprint, Temp Coefficient)

We did a complex analysis of passive part ratings, prices and stocking levels, and instituted some draconian policies on component selection.  This was based on both opportunities as well as expensive lessons.  _This GREATLY reduced design and NPI effort_.  All SMD resistors, and ceramic SMD capacitors were required to be OPEN_BRAND=yes, except if you had a really good reason, and these libraries we PRE-built with multiple man/pn fields for about 2-3 man each.  

Resistors default was 1%, 100ppm, and built with a very limited selection of standard values.  These were: (1, 2.2, 4.7) times 10, 100, 1K 10K, and 100K, with added 0, 50, 75, 150, and 2.49K.   Everything else was added as needed, as it would make dropdown lists too long, and part placement harder to put in all standard values.  Generally we used **only** 0603 resistors unless there was a voltage/wattage need.

SMD ceramic caps were a **huge research project**, and will be the subject of a video soon, as I think this market and its understanding is a MESS.  I'm sure there's a bunch of you that will barf about this, but the conclusion of our study was that you should use the smallest physical size part that meets your voltage and minimum capacitance spec (including C degradation with V).  There is negative spec advantage to any part at lower voltage than listed, and higher voltage parts are pushing price and availability issues.  You may need other values for timing circuits, and that's a whole different thing, but you should still spec the highest voltage of the smallest size high volume part you can find.  Even if you are only building stuff at 5V right now...  Anything else blows up the BOM for no reason.  **This is exactly the benefit of Kicost downloading the digikey database is all about.**

We settled on X7R default.  For each case size we created only 1-3 values, 0.1uF at its maximum commonly available voltage, and the maximum widely available capacitance included.  These were only about a dozen:
0402: 0.1uf 16V, 1uF at 10V
0603: 0.1uF 50V, 1uf at 25V, 0.01uf 100V
0805: 0.1uf 100V, 1uF 50V, 4.7uF 16V   
1206: nothing standard, good choices: 1uF 100V, 10uf 25V, 47uF 6.3V X5R, 1210: 10uF 50V 

 The important things here are the following issues/tensions:

1) Production people (especially CM's) DO NOT like choices.  They are extremely happy when there is only one part from one distributor to deal with, and they will happily charge you 10x the part cost to keep it that way.  They DONT CARE, and they DONT WANT to have to call in "engineering".  Alternates are the grudgingly accepted answer.
2) The cost of adding a line to a BOM, stocking the part, dealing with distributors, more alternates, etc is horrific, greatly exceeding the cost of almost all BOM parts.
3) The engineering cost of low availability specified parts is even higher.  Having to re-open a design and study it to deliver a new alternate sucks, and changing a part in one product can have terrible consequences.

From the point of view of KiCost,  there is a HUGE advantage to understanding and doing EXACTLY the hard part, of dealing with multiple MAN/PN specs, across multiple distys, with real part spec match.  Altium already has some of this capability, and the part database people are dealing with it every day.  The value of this is that you can push back against CM's really hard, on a large part of the BOM without expending tons of engineering time.  We are engineers, and we would really like to not spend our time scraping disty web sites for jelly bean parts.  

As a first step handling multiple MAN/PN fields would be awesome.