hildogjr/KiCost #18: Enhancement: Part Fields

stevenj, 2015-11-24:
At the moment the part fields are "manf#" and "newark#", "digikey#", "mouser#"

It would be nice if the supplier specific fields were documented in the docs on how to use KiCost.

As the program grows, I could imagine more custom fields being used.  Would it make sense to create some sort of field name space, so that other programs wont clash with the fields dedicated to kicost.  Something like "KiCost:manf#" instead of "manf#", etc.  It might also make the Schematic fields more self documenting, when someone other than the author see's those fields they have some idea of their purpose or at least when they google search "KiCost:manf#" it will probably turn up something useful.

stevenj, 2015-11-25:
Here is a proposal for a name space:

kicost:<option>:<option>:... = Value

instead of manf#:
kicost:manf# = <manufacturer part number> or "VALUE" 

The purpose of "VALUE" is to keep the schematic information DRY (Do Not Repeat Yourself) some components which are not replaceable, I like to put the actual manufacturer part number in the value field, so that its obvious what the selected part is.  This shouldn't have to be repeated in the kicost:manf# field so a token "VALUE" informs kicost to get the part number from the standard value field.  The "VALUE" token should work for any kicost component fields, even though it might not make sense for some.

kicost:dist:<newark# / digikey# / mouser# / anysupplier# > = <supplier catalog number> or "VALUE"

Add a special supplier called "manual" its used to provide manual pricing for parts kicost can not find.

eg:
kicost:dist:manual# = <manual part number>

Then to support this new fields:
kicost:manual:price = "<$.CC>" or "<qty:price>,<qty2:price2>,<qty3:price3>,..."

So one could supply just a unit price to be applied for any qty, or a qty price break list like a normal supplier.

kicost:manual:source = "URL where the price was obtained"

This field could be put into the spreadsheet or just used for documentation purposes.

To support generic components:

kicost:generic = "Generic specifier"

The Generic specifier is a string to help Kicost look up the generic component, its basic format would be something like"
<component type:param,param,param,param...>

The params would vary depending on the component, params would be name/value pairs that you are interested in.  such as V=33 (means value = 33)

The params would need to be different for each component type.

Examples:
0603 SMD 33Ohm Resistor
kicost:generic = "RES:V=33,P=0603"

This would find the cheapest 0603 33 Ohm resistor from each supplier in the required qty.

0603 SMD 33Ohm 1% Resistor
kicost:generic = "RES:V=VALUE,T=1%,P=0603"

This would look up the VALUE field in the component for the V (value) parameter.  Which in this example is 33Ohm, the Tolerance "T" is set to 1% and the Package "P" is set to 0603.  Its not possible to reliably use the Footprint field, because there is no guarantee that the package and footprint are the same names.  But in the spirit of DRY, could always put P=FOOTPRINT and it could be looked up.

Ignoring parts could be as simple as:
kicost:ignore=True

This would address issue of ignoring things like fiducials, mounting points, pcb edge connectors, test points, etc. Without resorting to using a heuristic which might or might not work accurately. 

Variant Part Lists:
I am working on a design at the moment with 7 variations based solely on component selection.  Pass a parameter to kicost called "variant" or something like that.  Then in addition to looking up these, also look up:

kicost:<variant>:ignore = "true"/"false"
kicost:<variant>:manf# = <manufacturer number>
kicost:<variant>:dist:<dist#> = <blah>
kicost:<variant>:generic:= <blah>
and so on.

If ANY "variant" field exists when that "variant" is selected it is used in preference to the base non variant fields, this way its possible to have a field which is ignored normally but in a particular variant is not ignored. Or vice versa, or in the normal case its a generic 33Ohm resistor but in the variant case its a specific part from Mouser. 

Using the name space allows easier extension, is easy to generate the required field keys pragmatically, is easy to read and maintain inside the schematic itself.

Can also emit errors if a field starting with "kicost:" is found but the rest of it is unable to be deciphered, so that the designer knows when they have made a mistake.

These are just suggestions, but i am working on a design with 7 variants, and I plan on enhancing kicost to deal with that, making a name space seems like the easiest way to manage the plethora of options that are useful for this kind of application.

xesscorp, 2015-12-30:
I implemented the "kicost:" prefix to distinguish KiCost-related fields from those of other tools. (The manf# field is still supported.) You can also add a label after that for a source or distributor. Then a final label indicating the type of data for the field's value (e.g., pricing). So, for example, the custom pricing field for a part might be:

    kicost:my_special_dist:pricing    =    1:$0.90; 10:$0.80; 100:$0.60

You can also have fields for cat# (the distributor's catalog number) and link (a link to the part's documentation). Parts can have multiple sources for each of these items.

You can also leave out the source label. This will imply a source of "local". So "kicost:pricing" would have a source of "local".

All the info for parts from a particular source is grouped into a section of the spreadsheet and labeled with the name of the source.

closed 2016-01-27 by xesscorp