hildogjr/KiCost #21: TypeError in Ubuntu 14.04

dale6998, 2015-12-16:
Hello,

I generate a BOM with bom2csv plugin.  This generates a valid xls file.
I run kicost with

` $ kicost -i test.xml -d 2`

and the output is:

`Get schematic XML...
Get parts library...
Get components...
Get groups of identical components...
Get parsed product page for each component group...
newark ['R4', 'R5', 'R1', 'R2', 'R3']
Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 9, in <module>
    load_entry_point('kicost==0.1.12', 'console_scripts', 'kicost')()
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.12-py2.7.egg/kicost/__main__.py", line 80, in main
    kicost(in_file=args.input, out_filename=args.output, debug_level=args.debug)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.12-py2.7.egg/kicost/kicost.py", line 100, in kicost
    list(distributors.keys()), part)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.12-py2.7.egg/kicost/kicost.py", line 1198, in get_part_html_trees
    fields['manf#'])
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.12-py2.7.egg/kicost/kicost.py", line 1456, in get_newark_part_html_tree
    tree = BeautifulSoup(html, 'lxml')
  File "build/bdist.linux-x86_64/egg/bs4/__init__.py", line 212, in __init__
  File "build/bdist.linux-x86_64/egg/bs4/builder/_lxml.py", line 109, in prepare_markup
  File "build/bdist.linux-x86_64/egg/bs4/dammit.py", line 263, in encodings
  File "build/bdist.linux-x86_64/egg/bs4/dammit.py", line 33, in chardet_dammit
  File "/usr/lib/python2.7/dist-packages/chardet/__init__.py", line 24, in detect
    u.feed(aBuf)
  File "/usr/lib/python2.7/dist-packages/chardet/universaldetector.py", line 115, in feed
    if prober.feed(aBuf) == constants.eFoundIt:
  File "/usr/lib/python2.7/dist-packages/chardet/charsetgroupprober.py", line 59, in feed
    st = prober.feed(aBuf)
  File "/usr/lib/python2.7/dist-packages/chardet/utf8prober.py", line 52, in feed
    codingState = self._mCodingSM.next_state(c)
  File "/usr/lib/python2.7/dist-packages/chardet/codingstatemachine.py", line 43, in next_state
    byteCls = self._mModel['classTable'][ord(c)]
TypeError: ord() expected string of length 1, but int found
`

Using Python 2.7 and installed with `easy_install kicost`

mminnella, 2015-12-21:
Same error here!,

seems that manf# that contains "-" exploit this error!

xesscorp, 2015-12-21:
I've never used the bom2csv plugin. All my tests have been done by generating the XML file using the BOM tool without any plugins enabled. Have you tried that?

It would help if I could see an XML file that demonstrates this problem.

mminnella, 2015-12-21:
Hello,

No i haven't tried yet , but i suppose that plugins simply parse xml file
and generate a new file (csv, or bom).
Anyway as attachment you could find my xml.

here the output for the command: kicost -d 6 --overwrite -i Receiver_1W.xml
-o Receiver_1W.xlsx

Get schematic XML...
>
> Get parts library...
>
> Get components...
>
> Get groups of identical components...
>
> Get parsed product page for each component group...
>
> newark ['C427']
>
> digikey ['C427']
>
> mouser ['C427']
>
> newark ['C401']
>
> digikey ['C401']
>
> mouser ['C401']
>
> newark ['R305', 'R304', 'R307']
>
> No 'newark#' field or 'manf#' field: cannot lookup part at newark
>
> Part not found at newark
>
> digikey ['R305', 'R304', 'R307']
>
> No 'digikey#' field or 'manf#' field: cannot lookup part at digikey
>
> Part not found at digikey
>
> mouser ['R305', 'R304', 'R307']
>
> No 'mouser#' field or 'manf#' field: cannot lookup part at mouser
>
> Part not found at mouser
>
> newark ['FB201', 'FB401', 'FB403', 'FB402']
>
> digikey ['FB201', 'FB401', 'FB403', 'FB402']
>
> mouser ['FB201', 'FB401', 'FB403', 'FB402']
>
> newark ['J301']
>
> No 'newark#' field or 'manf#' field: cannot lookup part at newark
>
> Part not found at newark
>
> digikey ['J301']
>
> No 'digikey#' field or 'manf#' field: cannot lookup part at digikey
>
> Part not found at digikey
>
> mouser ['J301']
>
> No 'mouser#' field or 'manf#' field: cannot lookup part at mouser
>
> Part not found at mouser
>
> newark ['C306']
>
> digikey ['C306']
>
> mouser ['C306']
>
> newark ['U302']
>
> Traceback (most recent call last):
>
>   File "/usr/local/bin/kicost", line 9, in <module>
>
>     load_entry_point('kicost==0.1.12', 'console_scripts', 'kicost')()
>
>   File
>> "/usr/local/lib/python2.7/dist-packages/kicost-0.1.12-py2.7.egg/kicost/__main__.py",
>> line 80, in main
>
>     kicost(in_file=args.input, out_filename=args.output,
>> debug_level=args.debug)
>
>   File
>> "/usr/local/lib/python2.7/dist-packages/kicost-0.1.12-py2.7.egg/kicost/kicost.py",
>> line 100, in kicost
>
>     list(distributors.keys()), part)
>
>   File
>> "/usr/local/lib/python2.7/dist-packages/kicost-0.1.12-py2.7.egg/kicost/kicost.py",
>> line 1198, in get_part_html_trees
>
>     fields['manf#'])
>
>   File
>> "/usr/local/lib/python2.7/dist-packages/kicost-0.1.12-py2.7.egg/kicost/kicost.py",
>> line 1456, in get_newark_part_html_tree
>
>     tree = BeautifulSoup(html, 'lxml')
>
>   File "build/bdist.linux-x86_64/egg/bs4/__init__.py", line 212, in
>> __init__
>
>   File "build/bdist.linux-x86_64/egg/bs4/builder/_lxml.py", line 109, in
>> prepare_markup
>
>   File "build/bdist.linux-x86_64/egg/bs4/dammit.py", line 263, in encodings
>
>   File "build/bdist.linux-x86_64/egg/bs4/dammit.py", line 33, in
>> chardet_dammit
>
>   File "/usr/lib/python2.7/dist-packages/chardet/__init__.py", line 24, in
>> detect
>
>     u.feed(aBuf)
>
>   File "/usr/lib/python2.7/dist-packages/chardet/universaldetector.py",
>> line 115, in feed
>
>     if prober.feed(aBuf) == constants.eFoundIt:
>
>   File "/usr/lib/python2.7/dist-packages/chardet/charsetgroupprober.py",
>> line 59, in feed
>
>     st = prober.feed(aBuf)
>
>   File "/usr/lib/python2.7/dist-packages/chardet/utf8prober.py", line 52,
>> in feed
>
>     codingState = self._mCodingSM.next_state(c)
>
>   File "/usr/lib/python2.7/dist-packages/chardet/codingstatemachine.py",
>> line 43, in next_state
>
>     byteCls = self._mModel['classTable'][ord(c)]
>
> TypeError: ord() expected string of length 1, but int found
>
>
Regards

Massimiliano

dale6998, 2015-12-21:
I believe you have to use a plugin in the 'new' KiCad (4.0).  Anyway, mminnella is correct...It just parses and produces a run-of-the-mill xls file.  Here is the script as it's called on by KiCad:

`

<!DOCTYPE xsl:stylesheet [
  <!ENTITY nl  "&#xd;&#xa;">    
]>

<xsl:stylesheet xmlns:xsl="http://www.w3.org/1999/XSL/Transform" version="1.0">
    <xsl:output method="text"/>

    
    <xsl:key name="headentr" match="field" use="@name"/>

    
    <xsl:template match="/export">
        <xsl:text>Reference, Value, Footprint, Datasheet</xsl:text>

            
            <xsl:for-each select="components/comp/fields/field[generate-id(.) = generate-id(key('headentr',@name)[1])]">
                <xsl:text>, </xsl:text>
                <xsl:value-of select="@name"/>
            </xsl:for-each>
            <xsl:text>&nl;</xsl:text>

        
        <xsl:apply-templates select="components/comp"/>
    </xsl:template>

    
    <xsl:template match="components/comp">
        <xsl:value-of select="@ref"/><xsl:text>,</xsl:text>
        <xsl:value-of select="value"/><xsl:text>,</xsl:text>
        <xsl:value-of select="footprint"/><xsl:text>,</xsl:text>
        <xsl:value-of select="datasheet"/>
        <xsl:apply-templates select="fields"/>
        <xsl:text>&nl;</xsl:text>
    </xsl:template>

    
    <xsl:template match="fields">

        
        <xsl:variable name="fieldvar" select="field"/>

        
        <xsl:for-each select="/export/components/comp/fields/field[generate-id(.) = generate-id(key('headentr',@name)[1])]">
            <xsl:variable name="allnames" select="@name"/>
            <xsl:text>,</xsl:text>

            
            <xsl:for-each select="$fieldvar">

                
                <xsl:if test="@name=$allnames">
                    
                    <xsl:value-of select="."/>
                </xsl:if>
                
            </xsl:for-each>
        </xsl:for-each>
    </xsl:template>

 </xsl:stylesheet>
`

mminnella, 2015-12-22:
Did you receive my xlm attached? 'cause as soon as i've receieved  the email, i just replied it with my xml, but i haven't found any trace of attachment here.

Regards,

Massimiliano

xesscorp, 2015-12-31:
Massimiliano, I didn't receive your XML attachment. Try putting it into a zip file before sending it. Thanks!

mminnella, 2015-12-31:
As attachment you could find my xml.

[Receiver_1W.xml.zip](https://github.com/xesscorp/KiCost/files/75617/Receiver_1W.xml.zip)

xesscorp, 2015-12-31:
I got the XML file. Thanks!

I ran it through KiCost (version 0.1.14). No errors were seen. I'm running Windows 7 Professional, 64-bit, and Python 3.4.3, 32-bit. What OS and Python are you using?

Can you try the latest KiCost version and send me a file with the debug output? (Make sure to use the "-s" option with the new KiCost so it uses a single process to do the web scraping. The debug output doesn't show up when parallel processes are used.)

mminnella, 2015-12-31:
Hello,
i've just update KiCost to KiCost 0.1.14.
my OS is: Ubuntu 14.04.3 LTS
my Python ver: 2.7.6 (default, Jun 22 2015, 17:58:13) 

here is the output from KiCost:

massi@wrk:/tmp$ kicost -s -d 6 --overwrite -i Receiver_1W.xml -o Receiver_1W.xlsx

Get schematic XML...
Get parts library...
Get components...
Get groups of identical components...
Create HTML page for parts with custom pricing...
<html>
  <body></body>
</html>
{u'digikey': {u'function': u'digikey',
              u'label': u'Digi-Key',
              u'order_cols': [u'purch', u'part_num', u'refs'],
              u'order_delimiter': u',',
              u'scrape': u'web'},
 u'mouser': {u'function': u'mouser',
             u'label': u'Mouser',
             u'order_cols': [u'part_num', u'purch', u'refs'],
             u'order_delimiter': u' ',
             u'scrape': u'web'},
 u'newark': {u'function': u'newark',
             u'label': u'Newark',
             u'order_cols': [u'part_num', u'purch', u'refs'],
             u'order_delimiter': u',',
             u'scrape': u'web'}}
Scrape part data for each component group...
newark ['C427']
digikey ['C427']
mouser ['C427']
newark ['R305', 'R304', 'R307']
digikey ['R305', 'R304', 'R307']
mouser ['R305', 'R304', 'R307']
newark ['L404', 'L406']
Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 9, in <module>
    load_entry_point('kicost==0.1.14', 'console_scripts', 'kicost')()
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.14-py2.7.egg/kicost/__main__.py", line 94, in main
    kicost(in_file=args.input, out_filename=args.output, serial=args.serial, debug_level=args.debug)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.14-py2.7.egg/kicost/kicost.py", line 129, in kicost
    id, url, part_num, price_tiers, qty_avail = scrape_part(args)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.14-py2.7.egg/kicost/kicost.py", line 1716, in scrape_part
    html_tree, url[d] = get_part_html_tree(part, d, distributor_dict, local_html)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.14-py2.7.egg/kicost/kicost.py", line 1691, in get_part_html_tree
    return get_dist_part_html_tree(dist, part.fields[key])
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.14-py2.7.egg/kicost/kicost.py", line 1606, in get_newark_part_html_tree
    tree = BeautifulSoup(html, 'lxml')
  File "build/bdist.linux-x86_64/egg/bs4/__init__.py", line 212, in __init__
  File "build/bdist.linux-x86_64/egg/bs4/builder/_lxml.py", line 109, in prepare_markup
  File "build/bdist.linux-x86_64/egg/bs4/dammit.py", line 263, in encodings
  File "build/bdist.linux-x86_64/egg/bs4/dammit.py", line 33, in chardet_dammit
  File "/usr/lib/python2.7/dist-packages/chardet/__init__.py", line 24, in detect
    u.feed(aBuf)
  File "/usr/lib/python2.7/dist-packages/chardet/universaldetector.py", line 115, in feed
    if prober.feed(aBuf) == constants.eFoundIt:
  File "/usr/lib/python2.7/dist-packages/chardet/charsetgroupprober.py", line 59, in feed
    st = prober.feed(aBuf)
  File "/usr/lib/python2.7/dist-packages/chardet/utf8prober.py", line 52, in feed
    codingState = self._mCodingSM.next_state(c)
  File "/usr/lib/python2.7/dist-packages/chardet/codingstatemachine.py", line 43, in next_state
    byteCls = self._mModel['classTable'][ord(c)]
TypeError: ord() expected string of length 1, but int found

tmanaud, 2016-01-06:
Same problem here: TypeError: ord() expected string of length 1, but int found
On Ubuntu 15.04, KiCost 0.1.14, Python 2.7.6, KiCad 4.0.0-rc2-stable
With an XML file generated with xsltproc -o "%O" "%I"
This issue arise when any manf# is filled, but not when if single distributor like mouser# is defined.

Looks like the issue is linux/xml related, how can I test it further?

tmanaud, 2016-02-13:
@xesscorp looks like you are running on Python 3 and us folk on linux install it with easy_install with python2.7 as recommended on the install page https://kicost.readthedocs.org/en/latest/installation.html

And after installation on python 3 with:
sudo easy_install3 kicost
Tjhis problem is solved.

metropt, 2016-04-22:
Use easy_install3 also solve the problem here.

xesscorp, 2016-04-24:
I replicated the error found by @dale6998 and @mminnella on Ubuntu 14.04. Here's how I fixed it:

1. apt-get install libxml2-dev libxslt1-dev python-dev zlib1g-dev

1. pip install -U kicost

The reason for the call to apt-get is to install the libraries that are needed to compile the lxml package used by kicost. I found using pre-compiled versions of lxml led to the problem with kicost. The problem went away once I rebuilt from source.

The other (and easier) alternative is to use easy_install3 as reported by @tmanaud and @metropt .

closed 2016-04-24 by xesscorp