hildogjr/KiCost #22: Out of stock part selected

cb88, 2015-12-16:
manf# = 5505207F
the script selects this part from mouser even though it isn't in stock. Digikey has it in stock and it gets a price for it... it just doesn't correctly highlight it as the least cost part.

cb88, 2015-12-16:
Similar problem for LTV-846S it finds the wrong page on digikey it finds LTV-846S-TA which isn't stocked but the part value I specified is stocked.

xesscorp, 2015-12-21:
KiCost doesn't select parts based on the quantity available, just the one with the lowest cost.

Send me the XML file with the LTV-846S in it and I'll try to see why it selects the wrong part.

cb88, 2015-12-21:
Here you go! This is not the original project I saw this problem in but it it still causes it.

[LedTest.zip](https://github.com/xesscorp/KiCost/files/69074/LedTest.zip)

xesscorp, 2015-12-31:
I fixed the problem where KiCost selects the wrong part from the list of available Digi-Key parts. KiCost was selecting based on Digi-Key's catalog number instead of the manufacturer's part number.

closed 2015-12-31 by xesscorp