hildogjr/KiCost #24: KeyError: u'manf#' at kicost.py 316 (version 0.1.14)

vooon, 2016-01-10:
I updated to latest and now program fails with following traceback:

```
Traceback (most recent call last):
  File "/home/vovan/.local/bin/kicost", line 9, in <module>
    load_entry_point('kicost==0.1.14', 'console_scripts', 'kicost')()
  File "/home/vovan/.local/lib/python2.7/site-packages/kicost-0.1.14-py2.7.egg/kicost/__main__.py", line 94, in main
    kicost(in_file=args.input, out_filename=args.output, serial=args.serial, debug_level=args.debug)
  File "/home/vovan/.local/lib/python2.7/site-packages/kicost-0.1.14-py2.7.egg/kicost/kicost.py", line 115, in kicost
    parts = get_part_groups(in_file)
  File "/home/vovan/.local/lib/python2.7/site-packages/kicost-0.1.14-py2.7.egg/kicost/kicost.py", line 316, in get_part_groups
    if components[ref]['manf#'] == manf_num:
KeyError: u'manf#'
zsh: exit 1     kicost -i testproject.xml
```

My fix:
```diff
--- kicost.py.old	2016-01-10 10:07:03.942035863 +0300
+++ kicost.py	2016-01-10 10:07:10.646094619 +0300
@@ -313,7 +313,7 @@
             sub_group.manf_nums = [manf_num]
             sub_group.refs = []
             for ref in grp.refs:
-                if components[ref]['manf#'] == manf_num:
+                if 'manf#' in components[ref] and components[ref]['manf#'] == manf_num:
                     sub_group.refs.append(ref)
             new_component_groups.append(sub_group)
             
```

xesscorp, 2016-01-10:
Hey, thanks for finding this bug!

I can't use your fix because it ignores components without a manf#. I actually need to enter those into the group with a manf# of None. So I used components[ref].get('manf#') which returns None if 'manf' isn't in the dictionary.

closed 2016-01-10 by xesscorp

vooon, 2016-01-10:
Good solution. My fix is "quick & dirty" :).

One last thing i found that now program starts too much threads (~30) so i have to use `nice -n10` else my old 2-gen i7 too abused.
And i sometime have problem with parts grouping, looks like adding `digikey#` somewhat affect merging.

xesscorp, 2016-01-11:
I added another enhancement to limit the number of parallel processes.