hildogjr/KiCost #31: PartNumber field in EEschema

sq5rix, 2016-02-14:
Hi,
I have just installed and run Kicost in Ubuntu 14.04 Kicad 4.01 stable and xsltproc. I used:
sudo easy_install3 kicost
It runs very well, I am able to get prices for my parts. 
Thank you very much for this software!!
I have noticed, that in schematic libs in version 4.01 there are default PartNumber fields in every standard library.
Would it take lots of work to be able to choose, whether to use manf# or PartNumber? Or maybe both, with manf# superseding PartNumber? 
It would save me lots of work to change the libraries one by one to add manf#
What do you think?

xesscorp, 2016-02-14:
Sounds like a good idea. I'll see what it will take.

sq5rix, 2016-02-14:
Maybe a command-line option 
--field PartNumber 
default would be manf# so legacy users would not complain

xesscorp, 2016-03-31:
I've installed the latest 4.0.2 stable version and I can't find any fields named "PartNumber" in the .lib, .sch, .net or .xml files for a design. Where are you seeing this field?

Also, unless the field shows up in the .xml file output by generating a BOM, KiCost won't see it. That's the only file it works with.

sq5rix, 2016-03-31:
Hi!

I have this field in every part - maybe try to get newest libraries from 
github? I use ubuntu 14.04 stable and kicad 4.02

Anyway, after some time of using kipart, I think it is not so important, 
I just add #manf, it takes one second and everything works ok.

I attach a screenshot.

Greetings,
Tom

metropt, 2016-04-22:
I also think that take this as an argument would be nice :)

hildogjr, 2017-12-28:
To some components, such as PICs, this code was as `Value`.

closed 2018-01-02 by hildogjr