hildogjr/KiCost #32: Unit cost per Economic Production Quantity (100 , 500, 1k, 3k ...)

tmanaud, 2016-02-16:
I propose the following improvement: the creation of an extra tab, in the output spreadsheet file, plotting economic production/order quantity: 100 , 500, 1k, 3k ...
It would greatly help builder to decide the batch size with a nice curve of price per unit and total investment.

hildogjr, 2017-08-30:
I did understand exactly, may you clarify me?
Is some way related with #51 ?
I think every propose that do us to minimize the production cost will be great. Fore example, #73 , because the ship to my country is not so cheap making conscious buy all the projects in the same distributor.

tmanaud, 2017-10-16:
I dont think this is related to #51, as this new feature would be a project production cost estimate and not componnent price.

hildogjr, 2017-10-16:
I think I understood.
For example, do you want new spreadsheet tab/page to include the PCB fotolito price, assembly cost and tax? And with this get a full production chain cost?
Write here your vision and what you expect of this new functionality.
Maybe this could be achieved in a new tab/page of the spreadsheet using a new module in the KiCost.

hildogjr, 2017-10-16:
A graph of the `Unit Cost` x `Board Qty`?
Maybe possible by a macro in Excel (and with will be did in a new page, I think is interesting add options do tax and fix cost). Could be a new module to be developed and add to KiCost as a separated `.py` file.
(E.g. in Excel/LibreOffice have the optimization tool that could allow you minimize one cell value sweeping other ones, it is used to production cost optimization. Of course, not the case of this issue.).

hildogjr, 2017-10-16:
https://stackoverflow.com/questions/1543562/excel-best-way-to-graph-the-change-of-a-cell-as-a-function-of-the-change-of-an
This may be the start point.

tmanaud, 2017-10-17:
Sure, it could be a start, on a extra tab on the current file would be perfect.
I think it's best to start with a very simple unit_cost/board_qty plot first, then add extra details such import fees, delivery, cut tape fees, etc. As this graph would be used only for decision taking when manufacturing in quantity, not prototyping.

hildogjr, 2017-10-17:
If the best way is to embedded a macro in the spreadsheet file, we need to do some manual test to make compatible, at least, with the the three suits (Microsoft Office, LibreOffice and Google SpreadSheet).

hildogjr, 2017-12-28:
Could something like this https://www.youtube.com/watch?v=S-W_eNsRiZ0 solve this issue?

hildogjr, 2018-02-24:
This feature will be not developed for now because different behavior between the three office suite supported.
The page still opened for discussions and ideas about this implementation.

closed 2018-03-10 by hildogjr