hildogjr/KiCost #34: Multiple BOM line items for a single reference designator

NF6X, 2016-03-19:
I propose a future enhancement, if it wouldn't be too cumbersome, to allow multiple line items to be added to the BOM spreadsheet based on a single component in the schematic. For example, a 2-pin jumper might add both a 2-pin header and a shunt. Or a mounting hole might add a screw, standoff and lockwasher. Or maybe I create a single component that represents something like an Arduino or an STM32F Discovery board plugged in as a daughtercard, with a custom footprint that has all of the various headers and mounting holes properly located, and I'd like multiple header strips or sockets to be added to the BOM based on that one component.

I think this would be a nice feature if there's a clean way to specify multiple part numbers with corresponding quantities in the fields of a component. I haven't thought much yet about how that might be cleanly accomplished, so maybe it would end up being too clumsy to be practical.

xesscorp, 2016-03-20:
This might be accomplished with a comma-separated list of part numbers in the manf# field. You would also need a corresponding number of entries in the manf field. Each part number could be prefixed with an optional multiplier. Then the webscraper could get the information on each part and send that back as a list to the other routines which would create a spreadsheet entry for each list item as is done now.

NF6X, 2016-03-20:
I like it. I think I vaguely recall seeing a pathological part number with a comma in it before, but if that problem ever crops up then some way to escape special characters could be added. For the quantity, a fractional number would allow for situations such as consuming 3 pins of a 36-pin breakable header strip.

hildogjr, 2017-08-24:
Yes, it is a good idea and the fractional part is important.
We may add in the interpretor or manf#:
PART -> normal part number
PART1;PART2 or PART1,PART2 or PART1, PART2 -> to list more than one part, in the spreadsheet the Refs, Value Desc, ... of the Global Part Info lines that reference to the same symbol may be merged.
1*PART1, 0.5*PART2, 3*PART3 -> to brought 1 of PART 1, 0.5 of PART2 and 3 of PART3 for each board (the total number will be ceiled / rounded to the next integer, because do not make sense buy 0.5 component).

hildogjr, 2017-08-29:
What you think @xesscorp and @NF6X? I think, before implement this (I could) it is important to discuss how will be this string and the definition of the multiplier symbol.
For example, I do not know with `*` is used in some manufecture / distributor number.

xesscorp, 2017-09-04:
I'm not sure about using a *; that's seems pretty common. Maybe you could use a delimiter like back-tick (`), ampersand (@), exclamation (!). It should probably also be allowable at either the beginning or end of the part number. Just set it up as a CONSTANT so that you can change it later.

I will amend my previous comment on how to implement this. Probably the place to do it is in the `get_part_groups()` function. That function could split the multiple part numbers and add the individual parts to the part list. Since there is only one part reference associated with multiple parts, it would have to append a sub-part index to the part reference. For example, if J2 consisted of a 2-pin header and a shunt, the header might have part reference J2-1 and the shunt would be J2-2.

hildogjr, 2017-09-04:
Interesting concept about the delimiter character. I think a valid discussion is use character that never is used as manufacture/distributor code, and also easy to type.

Yes, multiply first or after can be done (I will think how to implement). In this moment, we have to decide which character will be the multiplier and the break part allowing decimal / fractional. I was thinking:

    2/3 * PART1; 4.5 * PART2, 2*PART3; PART4   # PART4 is understood as multiplied by 1

    PART1 * 2/3 # Understanding also the post multiplier (I can implement)

Let's understand your propose:

    2/3 @ PART1! 3@PART2 ! PART#*7

?

I am working in an implementation, now that I understand better the code, yes, the implementation will be in `get_part_groups()`. I didn't think about add a "-subpart" (e.g. J2-1 and J2-2), and also it is a good and easy idea to implement.

hildogjr, 2017-09-07:
On the way.
I can now split the subparts in different items but, a can't use "." as delimiter as I told before because of some issue in the part of the code of `get_part_groups` function the put the list in group/order.

The partial code is in [https://github.com/hildogjr/KiCost/tree/multiparts](https://github.com/hildogjr/KiCost/tree/multiparts).

hildogjr, 2017-09-13:
My propose of request is working but do not recognize the sub quantities of the sub parts (identify but not add in the spreadsheet yet).
To do this, I need to modify the order routine and hash part in the main `kicost.py` file. Some one can help me?

hildogjr, 2017-09-21:
Working for KiCad XML BOM, but not for Altium yet. It is need to call the `subpart_split` function in the `altium.py`.

hildogjr, 2017-10-11:
Missing in this improvement: collapse equal subparts manufacture code in one.

Example, for different connector:

    J1 : ' MOLEX5PIN , 5 : 1PIN , 1/2 : CABLE6VIAS '
    J2 : ' MOLEX6PIN , 6 : 1PIN , 1/2 : CABLE6VIAS '

Should result at the spreadsheet in

    Ref                  Qty                     Manf#
    J1#1                1                        MOLEX5PIN
    J1#2,J2#2        ceil(5+6)                   1PIN
    J1#3,J2#3        ceil(1/2+1/2)               CABLE6VIAS
    J2#1                1                        MOLEX6PIN

I already predicted the `ceiling`function when get the quantity for subparts. Missing now change the collapse `collapse_refs()` collapse subparts for different footprints and sum its quantities (compose the string `1/2+1/2` in the example).

mdeweerd, 2017-11-27:
I just tried this feature.
I got multiple lines - one for each "manf#" part.
The "manf" field was not split though (both manufacturers appeared on each line).

hildogjr, 2017-11-27:
@mdeweerd, undertood. I can improve this and do a pull request.
So, you want (and make sense) that in `manf` be possible have different manufactures names in the same order of `manf#`. Correct? (It will be `,` or `;` separated as the `manf#`field standard).

mdeweerd, 2017-11-27:
Hello
Yes, this is what I mean.
You could allow an empty manufacturer name and imply that this repeats the previous name.

Example:

manf# = TI001,TI002,ST001,MOT001
manf= Texas Instruments,,STMicroelecronics,Motorola

Means that
TI001=Texas Instruments part
TI002=Texas Instruments part
ST001=STMicroelectronics part
MOT001=Motorola part

hildogjr, 2017-11-27:
@xesscorp, please just check the pull request #100 (correction about the multipart functionality). This improvement will be implemented there.

hildogjr, 2017-11-28:
@mdeweerd, I did the improvement. The files are in [my Multipart branch already pull requested](https://github.com/hildogjr/KiCost/tree/multiparts).

hildogjr, 2017-11-28:
@xesscorp, this issue discussion can be closed.

NF6X, 2017-11-30:
I'm sorry that I haven't been actively following this issue recently. I've just had too many other things on my plate.

Recent developments have forced me to update my KiCost installation, and I've found that using commas as subpart separators causes KiCost to crash when it encounters parts in my BOM whose manufacturer part numbers include commas. As discussed in issue #107, I've (temporarily?) worked around this by removing the comma from the PART_SEPRTR regex.

I don't know if that will cause other problems, because I do not yet have subparts in any of the BOMs I'm working with. I'm sure that I will make use of this feature in the future, though. Thank you to everybody contributing to this feature!

hildogjr, 2017-11-30:
@NF6X 
If you just change `PART_SEPRTR` from `[,;]` to `[;]` or `;` don't, do not create any problems and thanks to notice and propose a correction (I tested here and was OK, as in @mdeweerd  pull request #106 ).

closed 2017-12-27 by hildogjr