hildogjr/KiCost #38: Shows error for particular xml file alone

Arunmaran, 2016-06-03:
Hi All,
When i run attached particular file in the command for Kicost. It shows below error message. The file size is 175kb, is this a issue?
[BoulderCreekMotherBoard.zip](https://github.com/xesscorp/KiCost/files/297095/BoulderCreekMotherBoard.zip)

Error

C:\Python27\Scripts>kicost -i C:\Python27\Scripts\BoulderCreekMotherBoard.xml
Traceback (most recent call last):
  File "C:\Python27\Scripts\kicost-script.py", line 9, in <module>
    load_entry_point('kicost==0.1.19', 'console_scripts', 'kicost')()
  File "c:\python27\lib\site-packages\kicost\__main__.py", line 132, in main
    num_processes=num_processes)
  File "c:\python27\lib\site-packages\kicost\kicost.py", line 110, in kicost
    parts = get_part_groups(in_file, ignore_fields, variant)
  File "c:\python27\lib\site-packages\kicost\kicost.py", line 216, in get_part_g
roups
    for p in root.find('libparts').find_all('libpart'):
AttributeError: 'NoneType' object has no attribute 'find_all'

C:\Python27\Scripts>

xesscorp, 2016-06-03:
I ran your XML file through kicost 0.1.24 with no problems on Python 3 / Windows 7. With Python 2, the tqdm progress bar threw an exception at the end but it still generated the spreadsheet. So I'm not seeing any real problems with your file.

Make sure you're running the latest version. Also, enable the debugging by using the option "-d 3". That may tell us more about where the problem is.

The file you sent has no manf# fields in it, so kicost hasn't got any way to get cost data for the parts in your design. I'm not sure if that was intended. In any event, that didn't seem to cause a problem on my end.

Arunmaran, 2016-06-06:
I have updated the kicost and i am using python2.7 version. Still the same error as follows I also attached the file with manf# field.
[BoulderCreek_MotherBoard.zip](https://github.com/xesscorp/KiCost/files/300029/BoulderCreek_MotherBoard.zip)

Microsoft Windows [Version 6.1.7600]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Python27\Scripts>kicost -i C:\Python27\Scripts\BoulderCreek_MotherBoard.xml -
d 3
Get schematic XML...
Get parts library...
Traceback (most recent call last):
  File "C:\Python27\Scripts\kicost-script.py", line 9, in <module>
    load_entry_point('kicost==0.1.24', 'console_scripts', 'kicost')()
  File "c:\python27\lib\site-packages\kicost-0.1.24-py2.7.egg\kicost\__main__.py
", line 159, in main
    variant=args.variant, num_processes=num_processes)
  File "c:\python27\lib\site-packages\kicost-0.1.24-py2.7.egg\kicost\kicost.py",
 line 113, in kicost
    parts = get_part_groups(in_file, ignore_fields, variant)
  File "c:\python27\lib\site-packages\kicost-0.1.24-py2.7.egg\kicost\kicost.py",
 line 242, in get_part_groups
    for p in root.find('libparts').find_all('libpart'):
AttributeError: 'NoneType' object has no attribute 'find_all'

C:\Python27\Scripts>

Arunmaran, 2016-06-07:
Hi All,

Kicost is unable to process bigger file size say above 100kb.
Can anyone give some suggestion?

Or at least i need to generate proper BOM from saxonica.
At present it is generating as one component per line.
But i need as kicost format, means it should be grouped.

JoanTheSpark, 2016-06-13:
I can confirm the error (kicost 0.1.25 as per version info).
Does it matter if the Python 2.7 version (I got 2.7.2) is 32bit or 64bit?

xesscorp, 2016-06-13:
I'm using Python 2.7.10 and 3.4.3, both 32-bit. Are you using the same file that causes the error, or another one? Also, what version of lxml are you using? (I'm using 3.5.0.)

JoanTheSpark, 2016-06-13:
I used the file in the OP.
The python install was 2.7.2 32bit and kicost was 0.1.10 at the start and then I upgraded to 0.1.25
After I had trouble getting easy_install/pip running and also found that several python installs were/are crawling over each other on my system (EPD 2.7.3, a barebones 2.7.2, and some Panda3D python install) I went to de-install some, removed others from the PATH variable and in the end had a 2.7.11 64bit running (confirmed in command line) together with current lxml (think 3.6. something).
This is all on a Win 7 home pro 64bit.
I just don't know what else has been shot now by that kamikaze maneuver of mine.. :-(
Anyhow, the result was still the same as what Arunmaran got there.

xesscorp, 2016-06-13:
I've been running WinPython 2.7.10 and 3.4.3. They each keep all their files in separate directories and I have a small batch file that just puts one or the other first in the PATH variable on Win7.

I'll let this issue simmer for a while and go take a look at your other problem.

Mango-kid, 2016-06-27:
Same error for me (Ubuntu 14, python 3.4.3). Turns out it is the libxml packages need to be updated/added. Installing libxml2-dev & libxslt1-dev should so the trick.

closed 2017-09-04 by xesscorp