hildogjr/KiCost #40: Kicost for larger xml files

Arunmaran, 2016-06-07:
Hi All,

Kicost is unable to process bigger XML files size say above 100kb.
Can anyone give some suggestion?

Or at least i need to generate proper BOM from saxonica.
At present it is generating as one component per line.
But i need as kicost format, means it should be grouped.

xesscorp, 2016-06-07:
Based on your error trace, it might be caused by your version of lxml. I'm using version 3.5.0. You might try that if you're not already.

xesscorp, 2016-06-13:
Let me know the version of your lxml package and whether using 3.5.0 fixed this problem.

JoanTheSpark, 2016-06-13:
just upgraded python to 2.7.9 32bit, lxml is 3.6.0 (after installing pip, setuptools and wheels).
The xml file from the other bug report of the OP still brings the same error.

JoanTheSpark, 2016-06-13:
And now I'm running python 2.7.11 64bit, reinstalled lxml as there came up some bs4 error (bs4.FeatureNotFound: Couldn't find a tree builder with the features you requested: lxml) and still get the same error message.

PS: yeah, my own file doesn't work either with a different error.

closed 2017-09-04 by xesscorp