hildogjr/KiCost #41: main sheet without name.. error?

JoanTheSpark, 2016-06-13:
XML file goes like this:
```
<?xml version="1.0" encoding="UTF-8"?>
<export version="D">
  <design>
    <source>E:/Data_KiCAD/Projects/RPi-Test/RPi-Test.sch</source>
    <date>13/06/16 23:18:59</date>
    <tool>Eeschema (2016-03-04 BZR 6608, Git ba038ac)-product</tool>
    <sheet number="1" name="/" tstamps="/">
      <title_block>
        <title/>
        <company/>
        <rev/>
        <date/>
        <source>RPi-Test.sch</source>
        <comment number="1" value=""/>
        <comment number="2" value=""/>
        <comment number="3" value=""/>
        <comment number="4" value=""/>
      </title_block>
    </sheet>
    ....
</export>
```
And the error is then this:
```
Get schematic XML...
Get parts library...
Traceback (most recent call last):
  File "C:\Python27\Scripts\kicost-script.py", line 9, in <module>
    load_entry_point('kicost==0.1.25', 'console_scripts', 'kicost')()
  File "c:\python27\lib\site-packages\kicost-0.1.25-py2.7.egg\kicost\__main__.py", line 159, in main
    variant=args.variant, num_processes=num_processes)
  File "c:\python27\lib\site-packages\kicost-0.1.25-py2.7.egg\kicost\kicost.py", line 113, in kicost
    parts = get_part_groups(in_file, ignore_fields, variant)
  File "c:\python27\lib\site-packages\kicost-0.1.25-py2.7.egg\kicost\kicost.py", line 246, in get_part_groups
    fields = extract_fields(p, variant)
  File "c:\python27\lib\site-packages\kicost-0.1.25-py2.7.egg\kicost\kicost.py", line 207, in extract_fields
    name = str(f['name'].lower().strip())
  File "c:\python27\lib\site-packages\beautifulsoup4-4.4.1-py2.7.egg\bs4\element.py", line 958, in __getitem__
    return self.attrs[key]
KeyError: u'name'
```

To make this error vanish and get to the next step I have to add some string into the name field for sheet 1.
Which brings me to this error:
```
Get schematic XML...
Get parts library...
Get components...
Traceback (most recent call last):
  File "C:\Python27\Scripts\kicost-script.py", line 9, in <module>
    load_entry_point('kicost==0.1.25', 'console_scripts', 'kicost')()
  File "c:\python27\lib\site-packages\kicost-0.1.25-py2.7.egg\kicost\__main__.py", line 159, in main
    variant=args.variant, num_processes=num_processes)
  File "c:\python27\lib\site-packages\kicost-0.1.25-py2.7.egg\kicost\kicost.py", line 113, in kicost
    parts = get_part_groups(in_file, ignore_fields, variant)
  File "c:\python27\lib\site-packages\kicost-0.1.25-py2.7.egg\kicost\kicost.py", line 274, in get_part_groups
    fields = libparts[libpart].copy()  # Make a copy! Don't use reference!
KeyError: 'Power:VoltRef_MAX6034_2.5V_SOT323'
```
That part is the last in the libparts list and the 2nd in the components list.

xesscorp, 2016-06-13:
Can you attach the .xml file that shows these errors?

JoanTheSpark, 2016-06-14:
[RPi-Test.zip](https://github.com/xesscorp/KiCost/files/313244/RPi-Test.zip)

That's the file in question that stops with above error.

I made myself a little test project with 4 devices only that runs without a real problem (only the mentioned tqdm stuff that must be because one of the parts isn't defined for this kind of stuff).
This means Kicost is working as expected.

xesscorp, 2016-06-18:
Joan, your RPi-Test.xml file worked OK for me. No errors. I'm using kicost 0.1.25 with Python 3.

Let me know if you've seen other symptoms.

closed 2016-08-13 by xesscorp