hildogjr/KiCost #42: Support for Jameco vendor

mkdutchman, 2016-06-24:
I just started using kicost, but really like it already.

I also source some of my parts from Jameco, (www.jameco.com) would it be possible to use them with kicost? I'm not afraid of coding, and would be glad to help where needed.

xesscorp, 2016-06-25:
You could possibly make Jameco work with kicost. Here's what you need to do in kicost.py:

1. Put an entry with key `jameco` into the `distributors` dictionary. Set the `function` entry name to `jameco`. Set the `label` entry as `Jameco`. I don't know how to set the `order_cols` or `order_delimiter` since I don't know the format for Jameco order imports (or if it's even allowed).

2. Create a function called `get_jameco_part_html_tree`. This function searches for the page on the website with the given manufacturer part number or distributor catalog number. The page is parsed into a tree by BeautifulSoup and returned. The difficulty here is that the page returned by the website may not be the part page but a page with a list of possible pages for the part. Then you'll have to search that list to find the most likely matching page and return the tree for that. (The function for digikey does this.)

3. Create a function called `get_jameco_qty_avail` that looks in the page tree for the quantity of the part that's available. I don't see that the Jameco website gives this information, so maybe just return a default like `10000` or something.

4. Create a function called `get_jameco_part_num` that looks in the page tree for the manufacturer's part number and returns it as a string.

5. Create a function called `get_jameco_price_tiers` that looks in the page tree and returns a dictionary of quantities and associated prices.

Good luck!

closed 2016-08-13 by xesscorp

hildogjr, 2017-12-30:
@mkdutchman , opened an issue to discussions about new distributors support in #116.