hildogjr/KiCost #43: Run as a bom Plugin

stevenj, 2016-08-06:
Kicost can be installed into a virtual env and then run as a bom plugin.  Maybe could update the docs to include how to do it.

This is how i do it:
Create BOM plugin with

Name:
```
KiCost
```

Command Line:
```
"/usr/local/bin/venv-run.sh" kicost kicost -d 1 -i "%I"  -o "%O.xlsx" --overwrite
```

The script /usr/local/bin/venv-run.sh is as follows:
```
#!/bin/bash
# if virtualenvwrapper.sh is in your PATH (i.e. installed with pip)
#source `which virtualenvwrapper.sh`
source /usr/share/virtualenvwrapper/virtualenvwrapper.sh # if it's not in your PATH
workon $1
shift
"$@"
#python $2
deactivate
```

hildogjr, 2017-08-26:
Interesting tip and just putting `kicost - "%I"` works fine. But, of course, have the configuration as a plugin is better.
The only issue is that was better make this in a language recognized by Unix and Windows (because of part of the users), maybe bash is not the better option to this script (or make a DOS-command version for Windows). Also, leave the file in a global directory `/usr/share` may be not the better choice.

What about put this in `~/.local/lib/python**/site-packages/kicost` together with KiCost installation?

stevenj, 2017-08-27:
Feel free to make a Windows version of the script, I haven't used windows since Windows 98, and wouldn't know where to begin.  I don't even know if virtualenv runs on windows.

This is the sort of thing a user would likely want to tweak to suit their use case, which is why i think its better just to be documented, its relatively easy to manually set it up for anyone who wants it.

closed 2017-12-28 by hildogjr