hildogjr/KiCost #44: QTY Available 0

stevenj, 2016-08-07:
There are 2 reasons to use KiCost,

1 is for manufacturing NOW.  So i can buy the cheapest parts NOW to build a board.
Another is to get budgetary prices for your board.

In the first case, KiCost is great, as it removes any parts from a supplier with 0 Qty.  But in the second case, thats not very useful because we dont care if we can get the parts right now, we just care "if" we can get the parts whats the BOM cost.  Maybe the parts are on 2 week lead time and thats OK for you.

I propose we need a option that tells Kicad no to ignore parts that have 0 Qty Available, as long as the price is valid and the part was able to be scraped from a supplier.  Then KiCost can fulfil both roles.

stevenj, 2016-08-07:
Actually this looks like some kind of scraping error.  I am getting the Link but no other information for the parts in the worksheet, and the link shows those parts have ZERO qty available.

stevenj, 2016-08-07:
I hacked in the following test code.  The Parts are found whcih is why they have a Link, but there was no explanation given why they were being dropped, as there are 3 possibilities this code checks if the link exists, but the part is going to be dropped, which indicates a scraping error, and then expalins why the part was dropped. 

```python
 # If the part number doesn't exist or the part quantity is None
        # (not the same as 0), then the distributor doesn't stock this part
        # so leave this row blank.
        # Also leave this row blank if the pricing info doesn't exist so a $0
        # price isn't injected that messes up the search for the best price.
        if len(dist_part_num) == 0 or part.qty_avail[dist] is None or len(list(price_tiers.keys())) == 0:
            if part.url[dist]:
                logger.warning("Part %s scraping error from %s", part.refs, dist)
                if len(dist_part_num) == 0:
                    logger.warning("  -- No Distributor Part Number!")
                if part.qty_avail[dist] is None:
                    logger.warning("  -- No Quantity (Not Even Zero)!")
                if len(list(price_tiers.keys())) == 0:
                    logger.warning("  -- No Pricing Tiers!")

            row += 1  # Skip this row and go to the next.
            continue
```

 This is what it generates from my BOM:
```
Part ['R2'] scraping error from digikey
  -- No Quantity (Not Even Zero)!
Part ['R4'] scraping error from digikey
  -- No Quantity (Not Even Zero)!
Part ['R5'] scraping error from digikey
  -- No Quantity (Not Even Zero)!
Part ['D2'] scraping error from digikey
  -- No Quantity (Not Even Zero)!
Part ['U3'] scraping error from digikey
  -- No Quantity (Not Even Zero)!
Part ['D4'] scraping error from digikey
  -- No Quantity (Not Even Zero)!
Part ['JP1'] scraping error from digikey
  -- No Quantity (Not Even Zero)!
Part ['R1'] scraping error from digikey
  -- No Quantity (Not Even Zero)!
Part ['C44', 'C45', 'C47', 'C43', 'C46'] scraping error from digikey
  -- No Quantity (Not Even Zero)!
Part ['L1'] scraping error from digikey
  -- No Quantity (Not Even Zero)!
Part ['R2'] scraping error from mouser
  -- No Quantity (Not Even Zero)!
Part ['R5'] scraping error from mouser
  -- No Quantity (Not Even Zero)!
Part ['R1'] scraping error from mouser
  -- No Quantity (Not Even Zero)!
Part ['C44', 'C45', 'C47', 'C43', 'C46'] scraping error from mouser
  -- No Quantity (Not Even Zero)!
Part ['D2'] scraping error from newark
  -- No Quantity (Not Even Zero)!
Part ['P7', 'P10', 'P11', 'P8', 'P9'] scraping error from newark
  -- No Quantity (Not Even Zero)!
Part ['U4'] scraping error from newark
  -- No Quantity (Not Even Zero)!
Part ['U2'] scraping error from newark
  -- No Quantity (Not Even Zero)!
Part ['U7'] scraping error from newark
  -- No Quantity (Not Even Zero)!
Part ['L1'] scraping error from newark
  -- No Quantity (Not Even Zero)!
```

stevenj, 2016-08-07:
I added this test hack above the diagnostic i hacked in above.  Logging shows its triggered, but interestingly the spreadsheet now contains all valid data for those previously missing parts.  The Unit Price and catalog number show correctly.

I am not sure if this is the correct way to fix this otherwise I would submit it as a patch.
I inserted this at about line 1007:
```python
# Test Hack...
        if len(dist_part_num) != 0 and len(list(price_tiers.keys())) != 0 and part.qty_avail[dist] is None:
            logger.warning("Part %s correcting QTY to 0 for Distributor %s", part.refs, dist)
            part.qty_avail[dist] = 0
```

xesscorp, 2016-08-18:
Hi, Steven. While working on issue #45, I think I fixed the problem with only links appearing and no price/quantity for Digikey parts. It was related to changes Digikey made in their web page structure.

As for your original issue, KiCost currently chooses the lowest cost part without regard to availability, so it fits with your long-term, budgetary scenario. It doesn't fulfill the need for the short-term, buy-what-is-available-now scenario.

closed 2016-11-01 by xesscorp