hildogjr/KiCost #45: TypeError exception

NF6X, 2016-08-11:
Using an unknown older version of KiCost, I was getting zero quantity available for all items from Digi-Key, even for items in stock. I upgraded to the latest version of KiCost (0.1.27, as of 2016-08-11), and now it instead crashes with a TypeError exception after (I think) completing the scraping. Running with -d 5, here is the tail end of the console output:

```
running the remaining "atexit" finalizers
process exiting with exitcode 0
process exiting with exitcode 0
process exiting with exitcode 0
process exiting with exitcode 0
process exiting with exitcode 0
process exiting with exitcode 0
Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 9, in <module>
    load_entry_point('kicost==0.1.27', 'console_scripts', 'kicost')()
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.27-py2.7.egg/kicost/__main__.py", line 159, in main
    variant=args.variant, num_processes=num_processes)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.27-py2.7.egg/kicost/kicost.py", line 160, in kicost
    id, url, part_num, price_tiers, qty_avail = result.get()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 558, in get
    raise self._value
TypeError: ord() expected string of length 1, but int found
Exception TypeError: TypeError("'NoneType' object is not callable",) in  ignored
```

My initial attempts at debugging the issue have only resulted in pouting and confusion. I am open to suggestions for further debugging.

xesscorp, 2016-08-13:
Apologies for the problem. If you can send me the XML file and tell me the OS you're using, then I can try to find out what's going on.

NF6X, 2016-08-13:
It's an Ubuntu 14.04 LTS Linux box. Crash occurred with both .xml files I tried. Details and sample file to follow on Monday.

NF6X, 2016-08-15:
Here is a sample .xml file, with a log from running:

`kicost -i TestBoard.xml -o TestBoard.xlsx -w -s -d5 > & kicost.log`

uname -a output is:

`Linux mblair 3.13.0-93-generic #140-Ubuntu SMP Mon Jul 18 21:21:05 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

[TestBoard.zip](https://github.com/xesscorp/KiCost/files/419111/TestBoard.zip)

NF6X, 2016-08-15:
oliviermartin's patch linked to above appears to fix the crashing issue I'm having. After successfully failing to crash, I am having another issue: Some items are not showing availability at Digi-Key (not even a zero; just empty available and cost fields). These items receive links in the spreadsheet for Digi-Key pages of those items in reel quantity, with no reel available. These parts are still available in cut tape, but kicost isn't finding those parts.

Two examples:

Copal part number CL-SB-22A-01T
Wurth part number 431481031816

xesscorp, 2016-08-18:
I've found the solution for the blank Digikey fields in the spreadsheet. I'll clean it up tomorrow and post a new version.

commit 99fb3f1f (2016-08-18): Fixed scraping of Digi-Key pages to correctly detect reeled parts and scrape alternate packaging options. Closes #45, #46, #47.
* Bumped version number.
closed 2016-08-18 by xesscorp