hildogjr/KiCost #51: Add price break to each component as a comment in the Purch column

LeastSignificantBit, 2016-09-25:
As the title says, a comment that describes the different price breaks for that distributor would help deciding to buy in bulk or not on each component.

Digi-key has this function in their shopping cart, when hovering over a quantity you get a pop-up with e.g. 
Price Break 	Unit Price 	Extended Price
1 	                0,85000 	        0,85
10 	                0,50800 	        5,08
50 	                0,27580 	       13,79
100 	                0,23270 	        23,27

This would be great to have in the spreadsheet!

hildogjr, 2017-08-27:
I implemented this in `pull request`, a just have a new "issue". I want that the price get red color when is more cheaper to buy a quantity of the next price break instead of the actual quantity. I added this as a conditional format in `kycost.py`.

`
     test = '=iferror(and(if({purch_qty}="",{needed_qty},{purch_qty})<{max_break},{cel}>lookup(if({purch_qty}="",{needed_qty},{purch_qty}),{{{qtys}}},{{{prices}}})),"")'.format(
                    cel=xl_rowcol_to_cell(row, ext_price_col),
                    max_break=qtys[-1],
                    needed_qty=xl_rowcol_to_cell(row, part_qty_col),
                    purch_qty=xl_rowcol_to_cell(row, purch_qty_col),
                    qtys=','.join(qtys[:-1]), # Configuration to take the next price break value.
                    prices=','.join(prices_ext[1:]))

            wks.conditional_format(row, ext_price_col, row, ext_price_col, {
                'type': 'formula',
                'criteria': test,
                'format': wrk_formats['take_next_price_break']
            })
`

In a example, if I print the `test` variable returns `=iferror(lookup(if(K7="",G7,K7),{0,1,10,25,100,250,500,1000,2000,2500,5000,10000,25000,50000,100000},{0,0.43,0.304,0.2492,0.1994,0.14504,0.11784,0.092,0.088,0.08,0.072,0.0704,0.066,0.05852,0.0572}),"")` but inside of the spreadsheet got `IFERROR(AND(IF(K7="",G7,K7)<100,M7>LOOKUP(IF(K7="",G7,K7),{{qtys})))`.

If I type manually this conditional format in one cell (or copy and paste the printed string), works really fine. But did not work passing by `conditional_format(` and `.format(`. @xesscorp, may you help my if this issue?

hildogjr, 2017-08-29:
I implemented this improvement to show the price break as a balloon information in price and I was implementing a way to show that was more cheaper to buy more than the propose quantity (because of the price break) using the conditional format.

In the Xlxswrite forum I realize that just the LibreOffice allow ARRAY in conditional format, not the Excel (see the function of the column `Unit$`). So I don't know how to automate this.

Some tips?

hildogjr, 2017-12-28:
Closed, there is no easy way to inform this (break) in Excel, is the only the the three (Excel, LibreOffice Calc, Google Spreasheet) that not allow array in conditioning format.

closed 2017-12-28 by hildogjr