hildogjr/KiCost #52: Sheet name "too long"

mdeweerd, 2016-10-03:
The sheet name is based on the variants and as I divided the design into several options I got this error:
me
Exception: Excel worksheet name 'KiCost._rf_eth_usb_rtc_jack_header_' must be <= 31 chars.

I worked around this by having a fixed sheet name.

I think it is better to indicates the variants in the sheet's content itself.

Thank you.

hildogjr, 2017-10-16:
I checked and LibreOffice have no limitation, Google Spredasheet is <=100 and Excel <=31.
https://stackoverflow.com/questions/3681868/is-there-a-limit-on-an-excel-worksheets-name-length
Good idea, prioritize the `variant name` in the `spreadsheet name` to avoid errors.

mdeweerd, 2017-11-25:
I was just going to add an update to this thread - I will check the update.
If not, my workaround:
```
    if len(variant) > 0:
        # Append an indication of the variant to the worksheet title.
        # Remove any special characters that might be illegal in a
        # worksheet name since the variant might be a regular expression.
        variants = '.' + re.sub(
                                '[\[\]\\\/\|\?\*\:\(\)]','_',variant)
        if len(variants)+WORKSHEET_NAME < 31:
            WORKSHEET_NAME=WORKSHEET_NAME+variants
```

commit 9fe71001 (2017-11-28): Used some other branch codes

Included code to fix #52 and think in the #73 roadmap
closed 2017-11-28 by xesscorp