hildogjr/KiCost #53: Variants and "same" components - BOM incorrect

mdeweerd, 2016-10-05:
In order to build the right BOM and cost estimate for several variants, I have added fields like this (example for 100nF):

kicost.eth:manf#  -  GRM155R71A104KA01D
manf#  - GRM155R71A104KA01D
kicost.rtc:manf#  -  GRM155R71A104KA01D

Unfortunately, in some variant combinations all 100nF capacitors are added, and in some combinations, none of the 100nF capacitors are added, while in practice just a few should be added to the BOM in any combination.

hildogjr, 2017-12-28:
Could you add the XML and the kicost terminal commands that you are using?

mdeweerd, 2017-12-29:
This is an old ticket - I close it and I will submit an xml file the next file I observe the behavior.

closed 2017-12-29 by mdeweerd