hildogjr/KiCost #55: 0 Qty

466, 2016-10-24:
Hi there, 
i've just installed kicost using easy install, worked like a charm :)

I added manf# fields to all components in Esschema, created a BOM and used kicost on it.

Afterwards i do get an xlsx created with all parts needed, their footprints, manf# and stuff, but the quantity of each item remains zero. 
Which means each and every pcb does cost 0,00$, until i change the quantity of some items.
(So Kicost did get the price of each item) 

Any ideas on what i might have done wrong?

xesscorp, 2016-10-24:
I'll need to see your .xml file and the resulting spreadsheet to give any recommendations.

mazgis, 2016-10-28:
Hi, I'm facing the same issue but it looks like LibreOffice issue.
Then I select qty cell and press Function button then ok it starts working correctly.

466, 2016-10-30:
Yeha, sorry for the late reply.

Calculating the overall quantity via "=BoardQty***

closed 2016-11-01 by xesscorp