hildogjr/KiCost #59: Wrong product being returned from Mouser

cvgaviao, 2017-02-03:
Hi,
I've installed latest Kicost (0.1.31) from ubuntu ppa. I've run it using as source the default BOM xml exported by Kicad.
I'm facing an issue where KiCost is returning the wrong product from Mouser.
In the manfp# column in BOM I have B3U-1000P, but Kicost is returning this product: 653-B5T001001G from Mouser only.

xesscorp, 2017-02-17:
The latest version of KiCost (0.1.32) may have fixed this problem. Close this issue if so. Thanks.

cvgaviao, 2017-02-17:
great !
but how to update the version when I installed it via easy_install? 
I tried "sudo easy_install kicost", but:
````
    Searching for kicost
    Best match: kicost 0.1.31
    Processing kicost-0.1.31-py2.7.egg
    kicost 0.1.31 is already the active version in easy-install.pth
    Installing kicost script to /usr/local/bin
````

closed 2017-05-24 by xesscorp