hildogjr/KiCost #60: Problem "List index out of range"

srdjanStankovic, 2017-02-11:
HI,

I get a problem "List index out of range" in python.
[image]

**Description:**
KiCost 0.1.32
Win7 32-bit
Python 2.7

Any suggestions? Thanks

glcerone, 2017-02-12:
Hi,

I had the same problem that seems to be related to the Newark distributor, probably due to a website update. I didn't solved yet the problem but you can turn around it by using the --exclude option as follows in order to exclude the newark distributor from the search process:

kicost -i test.xml -e newark

this is not a solution but workaround the problem.

I have just opened a pull request to update kicost with the -e option that is not present in version 0.1.32. If you urgently need to produce the spreadsheet, you can download the code associated to the pull request.

best regards,

GL Cerone

xesscorp, 2017-02-17:
The latest version of KiCost (0.1.32) may have fixed this problem. Close this issue if so. Thanks.

antIggl, 2017-02-22:
The problem hasn't been fixed yet (commit 9569cd5a17e3d38cbafda402657b5ca988d9e558 ). The problem seems to be in subprocess' results parsing. I forked the repo and I ''ll have a glance on it.

glcerone, 2017-02-22:
Also for me the problem persists for Farnell and Newark distributors.

antIggl, 2017-02-22:
I ran a test for tests/test.xml, tests/test2.xml, and tests/test3.xml excluding farnell from the distributors search and it worked. Newark caused no problems for me. Could you confirm that the problem exist only for Farnell?

xesscorp, 2017-02-23:
I just uploaded a new version (0.1.33) that fixes the Farnell problem.

Kiogora, 2017-03-02:
Version 0.1.33 has this problem fixed. Just checked.

antIggl, 2017-03-02:
I 've just checked it too. It has been fixed on version 0.1.33.

closed 2017-05-24 by xesscorp