hildogjr/KiCost #63: Zeros in all lines of "Qty" column of output file "Global Part Info" section

antIggl, 2017-02-22:
Kicost does not parse or does not write the quantities for each part in the output file. The following screenshot shows the output file for tests/test2.xml. 
[image]

KiCost 0.1.32 ( commit 9569cd5a17e3d38cbafda402657b5ca988d9e558 )
python 2.7.13, Debian 9 (Stretch) GNU/Linux x86_64
libxml 20904 , libxslt 10129, libexslt 817

marelab, 2017-02-22:
Had a quite simalar problem, I had to Escape the Sheetpart of the NAMED fields. Its located in the kicost.py starting at Line: 671 down to Line :717
ORIGINAL -> wks_name=WORKSHEET_NAME,
CHANGED ->wks_name="'"+WORKSHEET_NAME+"'", 

after Escaping the 3 WORKSHEET_NAME vars it works as expected. Do not know if it has something to do with my Versions or maybe the german language.

WIN7
KiCost 0.1.32
LibreOffice 5.2 & OpenOffice 4.1.3

antIggl, 2017-02-23:
I forgot to mention that I use LibreOffice 5.2 (Version: 5.2.5.1 Build ID: 1:5.2.5-1). 
Well. The problem seems to be in the defined names, "BoardQty" and "TotalCost". The weird thing is that "Unit Cost" is calculated with "=TotalCost/BoardQty" and it works fine, but the "Qty" column for each part is defined as "=BoardQty*{part.refs}" and it doesn't works (on LibreOffice 5.2).
Will it be fine to replace those defined names on worksheet with cell references?

antIggl, 2017-02-23:
Update: Without any changes... forcing recalculation of "Qty" cells in LibreOffice works. It might be a LibreOffice problem.

xesscorp, 2017-02-23:
I uploaded a new version (0.1.33) that fixes the problem using the method given by @marelab .

antIggl, 2017-03-02:
It doesn't show quantities for each part, unless the field "Board Qty" gets updated (manually) on the sheet. It might be a problem of LibreOffice Calc.

closed 2017-05-24 by xesscorp