hildogjr/KiCost #64: UnicodeDecode Error

Gasman2014, 2017-03-20:
Running KiCost on macOS (in Conda virtual environment). Parser currently chokes on non-standard characters in the .xml file, specifically Ω, ±, % etc. These are all standard attributes in my Resistor library. Once I remove them, parser runs (but then falls over with another error). Looks like a utf8 problem?

hildogjr, 2017-12-27:
Could you attack one XML with this error?

Gasman2014, 2017-12-28:
To avoid this issue I have amended most of my libraries and now use a different strategy for my components (a database solution) but it is simple  enough to replicate - I can't attach a library but simply change the value of a resistor to say 10kΩ (i.e. using the Ohm sign) or add any of the other symbols mentioned to see the issue. (This BR was a while back...!)

hildogjr, 2017-12-28:
@Gasman2014, I just tested now with my code modifications (that will be merged to the main branch soon).
I got no error (Is this error persisting for you? Witch OS and KiCost version?).
I am using Ubuntu and my code is in: https://github.com/hildogjr/KiCost/
Please, close the issue with the issue is fixed (with this pull requested code).

closed 2017-12-29 by hildogjr