hildogjr/KiCost #68: ValueError: Attempted relative import in non-package

hexagon-robotics, 2017-05-23:
Hello,

I am trying to try KiCost but I get the following error.

`$ pwd`
.../KiCost/kicost

`$  python2.7 kicost.py `
Traceback (most recent call last):
  File "kicost.py", line 105, in <module>
    from .altium.altium import get_part_groups_altium
ValueError: Attempted relative import in non-package

I have installed the dependencies so in ubuntu xenial:
```
sudo apt-get install python-xlsxwriter
sudo pip install yattag
sudo pip install tqdm
```
Is another dependency necessary?

Regards,
Javier

xesscorp, 2017-05-24:
I'm using Python 2.7.12 on Windows 7 and not seeing any problem with importing.

timoalho, 2017-06-05:
I had the same problem on OS X 10.11.6, KiCost 0.1.35. For me it was fixed by running `python -m kicost ...`.

It's been a while since I've used KiCost, but I don't remember having this problem before, on a much older version of KiCost (and on a different OS X machine). I'm sorry I can't help you with more details right now, I'd have to dig up the old machine to check versions and so on.

hildogjr, 2017-12-27:
I installed KiCost by `pip` in Python3 and had no problem
`pip3 install --user kicost` To install to local user.
`sudo pip3 install kicost` To install to all users.
The maximum that a had to do before (one computer) was `sudo pip3 install --upgrade pip3`, all the other dependences was downloaded automatically.

closed 2017-12-28 by hildogjr