hildogjr/KiCost #69: Mouser Scraping

anderwm, 2017-05-24:
It seems like Mouser may not like to be scraped too quickly.  Or at least for me I get sporadic results unless I user serial scraping.  Anyway, you might check it out, and if I'm right put a limit on how fast you can do a "get" from them.

Sidenote: I'm blown away and grateful by how good of a job you have done on this project, it saves me a lot of hours.

xesscorp, 2017-05-25:
You're probably right about Mouser throttling their website. I have gotten partial data from them on one run and then gotten a full set immediately thereafter, but this behavior is sporadic. 

I'm glad to hear KiCost is doing you some good. Usually all I get are problem reports or requests for automatic currency conversion.

xesscorp, 2017-05-25:
I was able to reproduce your problem at 7 AM EST, but now it's vanished again at 12 PM. That's always been the frustration with solving this one.

I'll try again at various times and see if it's somehow related to time-of-day. Maybe Mouser has slow times and they do maintenance so throttling occurs. I don't know; I'm grasping at straws.

anderwm, 2017-05-25:
Yeah, it doesn't happen every time to me either, and it's not that big of a
deal big picture. I just thought some get limiting might help.

Having done some screen scraping before I know it is tricky to keep it all
working as distributors change their website. Between that and the way you
make the spreadsheet it is a really neat piece of code. Anybody that
doesn't appreciate that probably never did a high component count design
the old fashioned way.

If I get a chance I'll see if I can find a way to reliably reproduce the
issue and or solve it.

xesscorp, 2017-05-25:
Yeah, I'm getting the problem again. Maybe it's related to how much I've hit the site in a short period. I'll keep checking.

anderwm, 2017-05-25:
You are write, I ran it like this
` python -m kicost -i kicost/file.xml -d 99 --overwrite --include dist mouser`
a few times and started getting

> Exception while web-scraping

So then when I tried to browse to Mouser I got 

> Access Denied
You don't have permission to access "http://www.mouser.com/" on this server.
Reference #18.35eb2d17.1495740674.3e0b6f6

But then it lets me back on eventually.  So it is probably frequency of calls and then windowed over some time.

I added this to check the timing on line 1382(made a global like scraping progress):
```    # Scrape the part data from each distributor website or the local HTML.
    for d in distributor_dict:
        if d in dist_time_of_last_scrape:
            logger.warning("{} time since last scrape: {:.4f}".format(d, time.time() - dist_time_of_last_scrape[d]))
            time.sleep(1)
        dist_time_of_last_scrape[d] = time.time()
        # Get the HTML tree for the part.
        html_tree, url[d] = get_part_html_tree(part, d, distributor_dict, local_part_html, scrape_logger)```

closed 2017-09-04 by xesscorp