hildogjr/KiCost #70: Digikey all blank

sysrtime, 2017-05-27:
KiCost is a great tool!  But recently it's unable to retrieve parts from Digikey -- all parts come un "not found at digikey" but they're listed on Digikey's website.

On a related note: why scrape the website when Digikey offers a RESTful API for this purpose?

xesscorp, 2017-05-27:
Digikey may have changed their web page format. It was working a few days ago. I'll have to look.

Reasons not to use the [Digikey API](https://api-portal.digikey.com/start): rate limiting, charges for some plans, only provides price for a given quantity (can't get all the price break info at once), complicates setup with a unique token for each user, etc. I'd rather just scrape the site. Like a pirate.

xesscorp, 2017-05-27:
I just tested KiCost on Digi-Key. It's getting part information OK. Make sure you're using the latest version (`kicost -v` should print `0.1.35`). If you are, then send me the XML that shows the problem.

sysrtime, 2017-05-27:
Thank you for the prompt reply!!

I did some checking and discovered it works on Linux (Ubuntu 16.04 x64), but not on MacOSX (10.12.5, the latest).

Regards,
-david

closed 2017-06-05 by xesscorp