hildogjr/KiCost #71: Digikey scraping seems broken

timoalho, 2017-07-03:
No parts are found at Digikey, i.e. for all parts I get `Part ['mypartref'] not found at digikey`. To replicate, simply run any BOM including for example the part `20021321-00024T4LF`, which should be found but is not [(https://www.digikey.com/product-detail/en/amphenol-fci/20021321-00024T4LF/609-3774-ND/2209104)](https://www.digikey.com/product-detail/en/amphenol-fci/20021321-00024T4LF/609-3774-ND/2209104). The same seems to apply for any other Digikey part, this is just an example.

The same parts are found from the other distributors.

KiCost version 0.1.35, OS: OS X 10.11.6., Python 2.7

xesscorp, 2017-07-03:
I tried KiCost 0.1.35 on Python 2.7.12 with Win7 and it works fine on your example part and other test designs I have. I notice you're running OSX. Many of the problems that get reported are from that OS. I don't use a Mac so I don't have any insight into what the problems are.

Here are the versions of the libraries I'm using that KiCost needs. Maybe you can try to match that and see if the problem goes away:

BeautifulSoup4 == 4.6.0
XlsxWriter == 0.9.6
future == 0.16.0
lxml == 3.7.3
yattag == 1.8.0
tqdm == 4.11.2
CurrencyConverter == 0.13

timoalho, 2017-07-20:
I updated to python 3.6.2, and now it works. So it appears that the problem is specific to python 2 on OS X.

I'd venture a guess that this is because of some character encoding problem. However, since this works for me now, I unfortunately don't have the time/motivation to look into this further for the time being.

Package versions:
CurrencyConverter-0.13 
XlsxWriter-0.9.8 
beautifulsoup4-4.6.0 
future-0.16.0 
kicost-0.1.35 
lxml-3.8.0 
tqdm-4.14.0
yattag-1.8.0

closed 2017-09-04 by xesscorp