hildogjr/KiCost #76: Support for TME

maciekKuba, 2017-08-27:
I would like to add support for tme.eu distributor.

hildogjr, 2017-08-27:
This site understand the manufacture part number? (that identify a IC model + packge + work temperature ..., for example)

I did not know this distributor.

maciekKuba, 2017-08-27:
Yes it is. Here is an example 
http://www.tme.eu/en/details/ne5532p/tht-operational-amplifiers/texas-instruments/

adamheinrich, 2017-08-29:
Hi, I've recently started working on this feature, because I need TME as well:) It seems to be quite straightforward (copy existing distributor implementation and go through TME's HTML structure).

Have you already started? It would be a pity to duplicate the effort.

maciekKuba, 2017-08-29:
No, I didn't started. Ok, so how do you suggest to collaborate? I also noticed that TME delivers API. I looked on it and it seems to be an access to their database. Please look at this https://developers.tme.eu . Maybe it would be simpler to use it.

hildogjr, 2017-08-29:
Dear @adamheinrich,
What about merge this effort with the issue #77?

adamheinrich, 2017-08-29:
@maciekKuba I've just pushed my code from last week to https://github.com/adamheinrich/KiCost/tree/distributor-tme-dev. It seems to work, but it needs to be tested (and there are some TODO notes such as handling of all possible errors). Also, I made the code dependent on the `requests` module for AJAX requests; this might be redundant as some of the already used modules might be able to do the same thing.

The API requires registration and API key, which might be unwanted for this kind of application (every user would have to obtain her own API key as the key shall not be shared according to the documentation).

@hildogjr I'm affraid that this should be done by someone with deeper knowledge of both Python and KiCost internals...

hildogjr, 2017-08-29:
@maciekKuba , yes, I total agree, Because this I preferred propose in the forum and not in my pull request. I jut think that if they standardize all, make more easy to us add new module (if you see the `__main__.py` in the `rs`, `mouser`... folders, do not have function).

closed 2017-12-27 by hildogjr