hildogjr/KiCost #89: AttributeError: 'NoneType' object has no attribute 'group'

Dbroqua, 2017-10-11:
Hello,

When I try to use kicost on my project I have this error:

```Python
Progress: 100%
Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 11, in <module>
    load_entry_point('kicost==0.1.39', 'console_scripts', 'kicost')()
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.39-py2.7.egg/kicost/__main__.py", line 177, in main
    scrape_retries=args.retries)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.39-py2.7.egg/kicost/kicost.py", line 172, in kicost
    create_spreadsheet(parts, prj_info, out_filename, user_fields, variant)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.39-py2.7.egg/kicost/kicost.py", line 377, in create_spreadsheet
    wks, wrk_formats, START_ROW, START_COL, TOTAL_COST_ROW, parts, user_fields)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.39-py2.7.egg/kicost/kicost.py", line 656, in add_globals_to_worksheet
    part.collapsed_refs = ','.join(collapse_refs(part.refs))
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.39-py2.7.egg/kicost/kicost.py", line 499, in collapse_refs
    prefix = match.group('prefix')
AttributeError: 'NoneType' object has no attribute 'group'
```

In my XML all lines have not `manf#` value.

OS: Debian sid

Regards
Damien

xesscorp, 2017-10-11:
You'll have to attach your XML file.

Dbroqua, 2017-10-11:
You can find it here (sorry not linked in the first post >< ): https://pastebin.com/LcaS2gNH

whatnick, 2017-10-23:
Having the same issue in latest update. XML attached.
[din_meter_atm90e26.zip](https://github.com/xesscorp/KiCost/files/1406887/din_meter_atm90e26.zip)

hildogjr, 2017-12-27:
Error yet present in both files on the 0.1.42.

hildogjr, 2017-12-30:
Founded the source of the issue: @Dbroqua and @whatnick, both of you are using not allow (by KiCost) characters in the part references.
@whatnick : you have `_` in "WARN_1" and others.
@Dbroqua : you have `-`, `.`, `_`... in "6-6.25U_0" and other.

I am assigning the @xesscorp to check with the `PART_REF_REGEX` definition you be modified. But, for now, avoid this characters.

hildogjr, 2017-12-30:
I add some improvements and a error message about this. But I am assigning the @xesscorp to check if the `PART_REF_REGEX` definition will be modified.

For now @whatnick  and @Dbroqua, avoid this characters.

hildogjr, 2018-01-02:
@whatnick and @Dbroqua, the issue is fixed replacing `PART_REF_REGEX = re.compile('(?P<prefix>[a-z]+\W?)(?P<num>((?P<ref_num>\d+)({}(?P<subpart_num>\d+))?))'.format(SUB_SEPRTR), re.IGNORECASE)` in the `eda_tools.py` for

`PART_REF_REGEX = re.compile('(?P<prefix>[\da-z\.\_\-\+]+\W?)(?P<num>((?P<ref_num>\d+)({}(?P<subpart_num>\d+))?))'.format(SUB_SEPRTR), re.IGNORECASE)`

hildogjr, 2018-01-08:
Some EDAs (not common in KiCad, use sub number in the reference to group circuits by page, for example).

    "R0.1-R0.4,R1.1-R1.4,R2.1-R2.4,R3.1-R3.4"

For this work is necessary some additional modification in `PART_REF_REGEX` (Used in Altium and some others CSVs).

commit 13c8c9ad (2018-01-15): Fully fix #89 for even Altium BOM `PART_REF_REGEX`
commit 387cda49 (2018-01-15): Fix #89 #131 #133, improved #96, #130
commit d47c3314 (2018-01-15): Fully fix #89 even Altium BOM `PART_REF_REGEX`
commit df328660 (2018-01-15): Merge pull request #138 from hildogjr/master

Fix all forum bug (Fixed #89 #131 #133, improved #96, #130 )
closed 2018-01-13 by hildogjr