hildogjr/KiCost #95: other countries

melody8801, 2017-10-28:
# First: 
Thank you for your great work!!

## Then my Issue: 
is it possible to add the possibility to choose the country for the different different distributors? or ist that again a totaly new distributor? 

In my case, im looking for [https://www.digikey.ch](url) and if posible [https://www.distrelec.ch/](url).
im also willing to pay you a beer or two ;-)

Laogeodritt, 2017-11-01:
Hi there,

I think this is a duplicate of #65 that I submitted a while ago. Also linked to #12 and #56.

As a temporary workaround for you to be able to use KiCost locally, consider modifying the code to hard-code your country's domain names. Here's how I did it: https://github.com/Laogeodritt/KiCost/commit/4f2ce24291cb3b0ecf7a55b343f40b1fa9ecace6

(Note that this won't work for distrelec.ch as that's a totally different site - you'd have to write your own python module to query and parse their site!)

Regards,

Laogeodritt

hildogjr, 2017-11-02:
For https://www.distrelec.ch/ we don't have a scrap module. You could check #76 with the discussion of the last distributor module developed and start a issue exclusive to this development.
First of all, we have to check if the web site allow to use the manufacture code to search for a part/component.

melody8801, 2017-11-12:
Hi together

@Laogeodritt , thanks a lot to you. it did not eactly work like that, bud it worked like in my file descripted. (switzerland is allways a bit different with our three langauges and the possibility to choose different currencies). 

The code on line 145 should be 	
`url = 'https://www.digikey.ch/products/en?WT.z_header=search_go&lang=en&keywords=' + urlquote(				# Changed to Switzerland`

The code on line 149 should be
`url = 'https://www.digikey.ch' + url`
[digikey.txt](https://github.com/xesscorp/KiCost/files/1464701/digikey.txt)

@hildogjr : At first i am satisfied. I am currently at my "projectwork" at the BFH-Ti in Bern / Burgdorf and a do not have time to write a scrap module. But when my work ist done, i will write a module for Distrelec.ch. 

Yes it is possible to search for a manufacturer code. Works nearly the same way digikey works:
https://www.distrelec.ch/search?q=manuf#

hildogjr, 2017-12-28:
Close the issue.
Linked to internationalization and distrelec.ch add to "road map" in #116.

closed 2017-12-28 by hildogjr