hildogjr/KiCost #96: Processing of generic BOM

hallodeletue, 2017-11-09:
Hi,

first of all thank you very much for this great tool, I am using it on a regular basis for my diy audio hobby projects, and its an incredible boost in productivity, and a great money saver!

My proposal would be the ability to process generic BOMs, that are not necessarily created from a KiCad schematic, but a BOM in CVS or Excel format, containing only the most important fields, e.g. manufacturer number, customer number, order quantity. That would allow for a more efficient way of ordering BOMs for projects that you didn't design yourself or to quickly check costs of a project without drawing a schematic first.

Maybe there is already a way of achieving this that I am not aware of (some script to transform a generic BOM into a Kicad BOM XML), any hints appreciated.

Best regards,

hallodeletue

hildogjr, 2017-11-14:
Hi @hallodeletue, in the last pullrequest that I submitted to @xesscorp, I did some changes in the code thinking in this option. After this, will be necessary add a sub module to the `eda_tool` folder to make this read.
But yes, is possible but with some changes in the code.

hildogjr, 2017-11-14:
@hallodeletue, may you attach here some models of theses generic BOMs? This you help someone to understand better the requests.

hallodeletue, 2017-11-16:
Hi @hildogjr,

thanks for your comments. I attached an example of a generic BOM, including the colums
Mfr. No | Customer No | Order Qty. , I hope this helps. The one you attached is actually pretty close, 

It is of course also possible to have individual rows for unique parts, that is for example C1/2/3/4/7/8/13/14 don't get summed up in one row, but there are 8 rows in the table, if that is easier to parse. Then the colum "Order Qty" would be obsolete.

 However, I think it would be easier for "hand written" BOMS to have the field order quantity, and thus make KiCost a universal BOM checker.

Best regards,

hallodeletue
[generic_BOM_example.zip](https://github.com/xesscorp/KiCost/files/1480151/generic_BOM_example.zip)

hildogjr, 2017-11-18:
@hallodeletue, I did some implementation on [Hand BOM](https://github.com/hildogjr/KiCost/tree/hand_BOM) base on `altium.py` file.
Is not working yet, because this I not ask to merge. It is missing some better understand of the references codes / dictionary fields used by `KiCost`.
If you could help check or have some `altium.xml` to compare, it will be helpful. I made the changes needed at `main.py` and write a new sub module in `eda_tools`.

hildogjr, 2017-11-28:
@hallodeletue some minor issues (to specific CSV models) need to be fixed yet. But the [code](https://github.com/hildogjr/KiCost/tree/hand_BOM) is already working and pull requested.

hildogjr, 2017-12-13:
The KiCost 0.1.42 is getting the follow error with CSV files:

      File "/usr/local/bin/kicost", line 11, in <module>
        sys.exit(main())
      File "/usr/local/lib/python3.5/dist-packages/kicost/__main__.py", line 198, in main
        scrape_retries=args.retries)
      File "/usr/local/lib/python3.5/dist-packages/kicost/kicost.py", line 198, in kicost
        '-'.join(variant) if len(variant)>1 else variant[0])
      File "/usr/local/lib/python3.5/dist-packages/kicost/kicost.py", line 411, in create_spreadsheet
        wks, wrk_formats, START_ROW, START_COL, TOTAL_COST_ROW, parts, user_fields)
      File "/usr/local/lib/python3.5/dist-packages/kicost/kicost.py", line 702, in add_globals_to_worksheet
        part.collapsed_refs = ','.join(collapse_refs(part.refs))
      File "/usr/local/lib/python3.5/dist-packages/kicost/kicost.py", line 545, in collapse_refs
        prefix = match.group('prefix')
    AttributeError: 'NoneType' object has no attribute 'group'

Maybe some change needed in `PART_REF_REGEX` definition, @xesscorp? (I tested the CSVs of this discussion)

closed 2017-12-28 by hildogjr