hildogjr/KiCost #98: [Feature] Multiple manf#, manf for a part

mdeweerd, 2017-11-20:
Some parts can have alternatives from other manufacturers with different part number.

It would be nice to be able to specify this somehow (and get the pricing of course).

hildogjr, 2017-11-20:
@mdeweerd , maybe your request is already implemented and replicated from #34 ?
If yes (close this issue :-) ), check in documentation for "Parts With Subparts" (I coded this really to this function that you want, and for connectors assembly, using connector plastic + cable + X * pin).
Just fill the `manf#` in KiCad with `MANUFACTURER_CODE1 , MANUFACTURER_CODE2` or `1: MANUFACTURER_CODE1 , 1: MANUFACTURER_CODE2`
I think the documentation may be not clear about this use, so please, say us.

hildogjr, 2017-11-21:
Also, if you already know the part / component / manufacturer that you will use or want to do a different purchase list - price for each version. You can use the `variant` methodology (also in the documentation).

mdeweerd, 2017-11-25:
@hildogjr - Thank you for your feedback - I did see the #34 functionnality and I am using variants.

- variants are used to select different versions of the design (which components to mount to have this or that function);
- #34 is used for cases where a single reference in the design (example J3) requires multiple parts to be bought: a connector and its counter part, or different pieces composing the connector;

My request is different.  For example, I allow multiple references of a RJ45 MAGJACK.  Only one is needed.  The difference with #34 is that in this case only one item on the list is needed, while with #34 all items on the list are needed.

hildogjr, 2017-11-26:
I have used #34, enrolling the different `manf#` to get different lines in the spreadsheet. In the spreadsheet list, who will check, have to know the design to select just one to purchase.

mdeweerd, 2017-11-26:
I understand that #34 can be more or less used like this, but the spreadsheet would no longer be useable to get the "best" cost immediately as all components listed using #34 will appear as needed for the design.
So if you have two or three MAGJACK alternatives, the cost of the design would be at least two euros higher than the real cost.
Applying this to many components makes manual correction longer (and it would have to be repeated after every generation).

I guess that #34 is a step in the right direction.  However, the formulas in the spreadsheet should select the best alternative in the list (price/stock).

hildogjr, 2017-11-26:
Hum. Understood.
Maybe, already using #34 and with a additional flag (as the NOP that exclude the component from the spread sheet), could be write a formula to select the shipper option of than.
But to do this I think `kicost.py` `create_spreadsheet` have to be modified to include this functionality, maybe by some  Excel macro.

In some way, could be related with #51.

mdeweerd, 2017-11-26:
Hi

If using #34 functionnality, one could effectively indicate in another field if this is a single component or not.

I think it could be done without a macro.

The following code needs a change to refer to the N-1 next rows if these are alternatives to find the best price,
```
        # Enter the spreadsheet formula to find this part's minimum unit price across all distributors.
        wks.write_formula(
            row, start_col + columns['unit_price']['col'],
            '=MINA({})'.format(','.join(dist_unit_prices)),
            wrk_formats['currency']
        )
```

and change of 'xl_rowcol_to_cell' .
Based on the value for 'row' and prior analysis, one can determine if row it part of a list of alternatives or not.

hildogjr, 2017-11-26:
Interesting.
Maybe defining `SHIPPER`=1 (as we do with `NOPOP`), this peace of code could be write for this group of `manf#` codes.
I see that two actions are needed (if will use #34):
1) Set the shipper to be purchased;
2) Set the other rows (correspondent to other codes) to 0 quantity.

We have to worrying that this have to dynamic with the component/board quantity. Because, the most shipper component could change with the quantity.

mdeweerd, 2017-11-26:
I would suggest to:
- Keep the quantity and price on the first line.
- To remind in the next lines (in the description) that this is the above line continued (using sequence of quotes for example """"""""" );
- The first line needs to select the best price from the N lines (if N alternatives);
- The distributor conditional formatting needs to compare with the first line.

That avoids having to set quantities to 0 - they will already be 0.

Which is why I suggest:
a. A table for xl_rowcol_to_cell to know what the reference row is for the given "row".
b. 'wks.write_formula' for the minimum price would select minimum price of multiple rows and could be the MINA of several MINA results (one for each row).

hildogjr, 2017-11-26:
Could it be a solution. Even the actual multi parts could be improved in this way.
I see some interesting use, some times I buy 7805 for example, not concerning about temperature e so on, just price and stock quantity.
Just we have to be concerned about it will work when the parts will be grouped (more than one designator will follow this rule of multiple `manf#`). Or when some designator follow this rule and order one don't, taking just one of the enrolled `manf#` as valid (how to group in the spreadsheet?).

hildogjr, 2017-11-26:
Other thing to think, when different `manf#` is the chippest across distributors.
Just to enroll here the difficulties in the low level language and how to solve.

mdeweerd, 2017-11-26:
I created an excel file with two lines that are supposed to represent the same component type.
For Qty of 1, the first line has the lowest cost, for Qty of 100 the second line is the lowest cost.

[ExampleMultiple.xlsx](https://github.com/xesscorp/KiCost/files/1504188/ExampleMultiple.xlsx)

I added conditional format on column "Manf#" for lines 7 and 8 (comparing the minimum value on the line with the unit$ as it is done for the distributors more or less).  Column 7 could use a similar formatting.
The Unit$ on line 7 checks the prices on lines 7 and 8.
The distributor's conditional formulas compare to G7 and G8.

This would already be nice.

To improve, the distributors should depend on other value than Qty column and the Qty column would need to be 0 if the line is not the lowest price.
To avoid circular references, we could have an extra Qty/board column (which is what the manufacturers need) and the distributor columns can then do Board Qty x Qty/board.  Then in column Qty we can put the actual Qty for the component.

hildogjr, 2017-11-26:
I think to be clear to the user, we need implement a white-gray alternating rows background (quite easy) or border.

mdeweerd, 2017-11-26:
That can help - to be able to copy/paste we need to have the Qty/Board in a column and total Qty in another.

mdeweerd, 2017-11-29:
Regarding our discussion, here is a suggested read:
https://rheingoldheavy.com/kicad-bom-management-part-1-detailed-components/

That user suggests MFN and MFP (kicost fields are "manf" and "manf#"), and then "S1PN,S1FP", "S2PN,S2FP" and we could imagine to have 3,4, etc.

hildogjr, 2017-12-30:
How about create the two rows in the spreadsheet but joined by one border, symbolizing (to the user) so kind of group. The `qty` have to a merged cell, indicating the the `qty` refers to both and, the price (for now) we get as the minimum of the 2 items. By my KiCost code acknowledgement appear more easy than add a "thirty title column".
In the EEschema component part we define as `ALIAS` (such the `NOPOP` definition) to indicate the the `manf#` codes refers to the same components (the user in the spread sheet will be induced to the shipper, or shipper first until the maximum quantities).
The behavior of the `ALIAS` "flag" in EEschema will be:
1. If just defined, all `manf#` refers to the same component;
2. If use number "inside" the length of `manf#`, it will indicate each one is alias and not. E.g:
 
`manf#` = HEAT_SINK, ISOLATOR, SCREW, MOSFET1, MOSFET2, MOSTEF3
`ALIAS`= 4,5,6   or 4:6  or 4:

"4:" is the Python indication 4 to end.

mdeweerd, 2017-12-31:
I think that the "alias" proposition may be usefull to some, but this  is not what I mean.
Here is an example:
[image]

I specified the alternatives as S1*, S2* (choice based on a blog article I referenced in another "issue").

The "smaconn" variant can have on of three different components, depending on the cost: "Hope RF/SMA EB", "RF Solutions/CON-SMA-EDGE" or "Multicomp/19-70-1-TGG".
Here one would have to scrape the costs for the 3 types, and select the best one based on Qty/Cost.

I just worked with a manufacturer that added an extra column to the BOM which is "Selected reference" followed by the cost for that component and columns with pricing for different selected quantities.
I like that ;-):
[image]

https://github.com/xesscorp/KiCost/issues/113 adds another level to that: the possibility to chose between variants with different manufacturer possibilities.

hildogjr, 2017-12-31:
In KiCost's generated spreadsheet, this price by quantity (price breaks) are display as a balloon tip at the cell unit price for each distributors scraped and are automatically selected by the quantity purchased.
This column "Selected reference", I did understand how it works, could you attack same example spreadsheet?

mdeweerd, 2018-01-01:
Here is an example sheet combining some of my proposals:
- Distributor parts moved to seperate sheet;
- Multiple references from same distributor possible for same reference (I did not show all the references on the main sheet, but that is easy);
- Multiple lines possible for same manf# while still taking into account global cost (should be option to have one line or multiple lines);
- Selected manf# changes according to quantity;
- Manual updates to quantities (board qty or qty per board) is possible;
- Three default quantities to get quick idea of cost according to quantity without manual changes.

[ExampleMultiplePartTypes.xlsx](https://github.com/xesscorp/KiCost/files/1595550/ExampleMultiplePartTypes.xlsx)

hildogjr, 2018-01-11:
Just to keep the ideas written, better use (){}... in man#:

    manf# = HEAT_SINK,{ISOLATOR1, ISOLATOR2}, SCREW, {MOSFET1, MOSFET2, MOSTEF3}

So, components are:

    HEAT_SINK
    ISOLATOR1 or ISOLATOR2
    SCREW
    MOSFET1 or MOSFET2 or MOSTEF3

hildogjr, 2018-02-24:
This new feature, possible, demand to rewrite the `spreadsheet.py` file. So, don't have a plan of implementation.

closed 2018-03-10 by hildogjr