hildogjr/KiCost #102: Ignore "desc" and "variant" for joining Part info, but keep them for selecting and column values

mdeweerd, 2017-11-26:
Use case:
I have several variants but same components in the variants.
Example: 100nF with same footprint across variants.

I want to ignore the description field (desc) and variant field for joining the references in the table, but the 'desc' field should still appear in the table and the 'variant' field should still be available for selecting the parts.

However, currently the ignored fields are not read.  Therefore, when ignoring the description, the "Desc" field is empty in the output, and when ignoring the 'variant' field, I think that the variant selection no longer works.

So the question is that for special fields like "desc" and "variant" that the values are read and  shown in the excel output (for the desc field) but ignored for grouping and continue to be used for selecting (like the variant field).

hildogjr, 2017-11-26:
This issue also affects/is the cause of the #97, #73 (in pull requests / been developed by me) and don't allow group equal sub part of a multiple part designator (for example when is purchased cable or pib to different connectors). Or even if you use "pin to test points" and call than by the signal name to used in the board silk.

I think this will need some rewriting of the group way and is a recurrent issue.

mdeweerd, 2017-11-30:
I've implemented something in my "fork" which seems functionnal now (for kicad):
https://github.com/mdeweerd/KiCost/commit/e351069ea82ef94ba5e84f63bed3202aa8000837

To use it, the desc and/or variant field has to be added to the list of fields to ignore.  This then triggers the functionnality.
The desc and variant fields are first removed fromt the ignore list in the kicad code in order to retrieve the values of the fields, but then they are ignored when computing the hashing.

I still ran in to issues as I've also started adding second sources using extra fields, but that will need extra thought (these fields should be handled as manf data and therefore also ignored like the manf and manf# fields).

mdeweerd, 2017-12-06:
I just pulled the latest version - the refactoring of the grouping breaks my code :-(.

hildogjr, 2017-12-30:
The ideia may be:
- Ignore `desc`, `variant` and maybe `footrpint` (case related with #97 and #73, maybe by a parameter in KiCost call).
- Keep this information to control in the spreadsheet, probably merging the strings with `;` (e.g. `desc_spreadsheet` = `DESC_PART1; DESC_PART2`). Maybe, to control add a _conditioning format yellow_ in this cell to user visualization of a "special group/merge".

mdeweerd, 2017-12-31:
- The fields to be ignored when joining could definitively be specified in a parameter as suggested.  And/or let the user specify the fields to use for joining (rather than specifying which ones not to use).
- When joining descriptions, etc: add the references in front of it.   So "R1:DESC_PART1;R2-R3:DESC_PART2"
- Let user choose if empty field should be same as non-empty field or not (suppose USESAME).
  - Unique description + others empty.
     If USESAME, then empty field = unique description, otherwise empty.
  - Multiple descriptions + others empty. 
     Option does not apply, empty field is always empty (can not choose from multiple choice).

Other suggestion(s)/thoughts:
- It might be easier to use the Python script only to scrape the commercial data, and implement the other logic in the spreadsheet.  The spreadsheet would have a sheet with the components and their costs, a sheet that would look a lot like the original BOM, and "the current sheet" that could be built using macros, etc.  I think that variant selection could be pushed to the spreadsheet.
- The script could use a template .odt or .xls(m) to add the data to which allows to develop the macros independently from the python script, but also allows the end user to build his own template.

hildogjr, 2017-12-31:
Macros may be a problem because LibreOffice Calc / Google Spreadsheets / MS Excel have different languages to macro / definitions, keep the compatibility with the 3 along the new features in KiCost is an important job (#35). I think, because this, KiCost will not assume this direction, for now  (conceptual tests have to be made, but macro could be a way to solve #32 for example). Even, every time, allow the macros to be activated could be confuse to not experts user. 

About the field to be ignored, by default (my experience, but I am waiting @xesscorp opnion) `desc`, `variant` and some generics have to be ignored in to group. For sure, `manf#` is the most important, but sometimes even `manf`, `footprint` (because user library names definitions along the softwares) is not important to group distinguish.

Maybe `kicost --ignore_group desc` call could be created. By default `ignore_group = ['desc', 'variant', 'manf']`.

@mdeweerd, interesting idea about "R1:DESC_PART1;R2-R3:DESC_PART2".

hildogjr, 2018-01-19:
Implemented, use `kicost -i %file% --group_fields desc`for group the different description, These are kept in the spreadsheet as different lines in the respective "desc" cell.

hildogjr, 2018-03-16:
Reopened as automatic merge the `desc` and `variant` in the spreadsheet. Use the code already existent passing as parameters this fields.

commit e825b9fc (2018-01-19): Merge pull request #158 from hildogjr/master

Fixed #102 #156 and started #149
commit c476f703 (2018-03-17): Complety fixed #102
closed 2018-03-17 by hildogjr