hildogjr/KiCost #106: [Bug] Result not generated

mdeweerd, 2017-11-29:
I go the following error after everything was scraped:
```
Traceback (most recent call last):
  File "/usr/bin/kicost", line 9, in <module>
    load_entry_point('kicost==0.1.41', 'console_scripts', 'kicost')()
  File "/usr/lib/python2.7/site-packages/kicost-0.1.41-py2.7.egg/kicost/__main__.py", line 194, in main
    scrape_retries=args.retries)
  File "/usr/lib/python2.7/site-packages/kicost-0.1.41-py2.7.egg/kicost/kicost.py", line 182, in kicost
    id, url, part_num, price_tiers, qty_avail = result.get()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 567, in get
    raise self._value
TypeError: unhashable type: 'list'
Exception TypeError: TypeError("'NoneType' object is not callable",) in <bound method tqdm.__del__ of Progress:  98%|############################################################################################4 | 58/59 [00:00<00:09,  9.09s/part]> ignored
mv: cannot stat 'BOM_15288proto.xlsx': No such file or directory
```

I know that it is not much, but maybe you have a hint where to start looking?

hildogjr, 2017-12-06:
At last version of KiCost something wrong is happening, using the XML of #112 , the process still freeze in 0%. Even debugging, no part return as scraped.
@xesscorp, this started fill commits ago.

mdeweerd, 2017-12-06:
It think that it is usefull to "cache" scraped information in a local xml file.  Through an option the user of kicost could either ignore the cache, define the default maximum age of the cache information, etc.
It could be done using a "caching proxy" internal to kicost which could read/fill an sqlite database or xml file using the same format as the existing virtual "local" pricing information.

I am not sure that this is the solution for this issue though.

hildogjr, 2017-12-07:
Testing a little more, my error is related with the Farnell scrap module  / web site. The newark is also slow but just slow.
To work around this issue I am using `--include digikey mouser rs tme newark` or `--exclude farnell`. So the process not freeze in 0% anymore.

closed 2017-12-27 by hildogjr