hildogjr/KiCost #107: Comma in Manufacturer Part Number

NF6X, 2017-11-29:
KiCost is crashing for me, and the trigger appears to be a component in my BOM which has a comma in its manufacturer part number (Nexperia part number BSS138PW,115). If I remove the manf# field from this component in the XML file, then KiCost no longer crashes, but it fails to find the part at Digi-Key based on the abbreviated Value field of BSS138PW. I suspect that the comma in the manufacturer part number is breaking the parsing in the scraper?

Here is the console output:

```
Get schematic XML...
Get parts library...
Get components...
Search for subparts within parts...
('Q20', '>>', ['manf#'])
('part/qty>>', u'BSS138PW', '\t\tpart>>', u'BSS138PW', '\tqty>>', '1')
{u'libpart': 'Transistors:BSS138PW', 'manf#_subqty': '1', u'value': 'BSS138PW - p1/2', u'manf#': u'BSS138PW', 'manf': 'Nexperia', u'footprint': 'Transistors:SC-70'}
('part/qty>>', u'115', '\t\tpart>>', u'115', '\tqty>>', '1')
{u'libpart': 'Transistors:BSS138PW', 'manf#_subqty': '1', u'value': 'BSS138PW - p2/2', u'manf#': u'115', 'manf': 'Nexperia', u'footprint': 'Transistors:SC-70'}
Get groups of identical components...
Create HTML page for parts with custom pricing...
<html>
  <body></body>
</html>
{'digikey': {'label': 'Digi-Key',
             'module': 'digikey',
             'order_cols': ['purch', 'part_num', 'refs'],
             'order_delimiter': ',',
             'scrape': 'web',
             'wrk_hdr_format': {'align': 'center',
                                'bg_color': '#CC0000',
                                'bold': True,
                                'font_color': 'white',
                                'font_size': 14,
                                'valign': 'vcenter'}}}
Scrape part data for each component group...

Progress:   0%|          | 0/2 [00:00<?, ?part/s]digikey ['Q20#2']
Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 9, in <module>
    load_entry_point('kicost==0.1.41', 'console_scripts', 'kicost')()
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.41-py2.7.egg/kicost/__main__.py", line 194, in main
    scrape_retries=args.retries)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.41-py2.7.egg/kicost/kicost.py", line 151, in kicost
    id, url, part_num, price_tiers, qty_avail = scrape_part(args)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.41-py2.7.egg/kicost/kicost.py", line 1263, in scrape_part
    html_tree, url[d] = get_part_html_tree(part, d, dist_module.get_part_html_tree, local_part_html, scrape_retries, scrape_logger)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.41-py2.7.egg/kicost/kicost.py", line 1211, in get_part_html_tree
    for extra_search_terms in set([part.fields.get('manf', ''), '']):
TypeError: unhashable type: 'list'
Exception TypeError: TypeError("'NoneType' object is not callable",) in <bound method tqdm.__del__ of Progress:   0%|          | 0/2 [00:00<?, ?part/s]> ignored

```

NF6X, 2017-11-30:
Looking into this further, I found that support for multiple BOM line items per schematic component are now supported (yay!), but that commas are accepted as subpart separators in the manf# field (boo!). This causes problems in the BOM for the design I'm about to release, because it includes two components (Nexperia BSS138PW,115 and Toshiba SSM3J328R,LF) with commas in their manufacturer part numbers.

**I propose that commas should not be be accepted as subpart separators in the manf# field, since they can legitimately appear in manufacturer part numbers.** I'll create a pull request which removes the comma from PART_SEPRTR and adds a test case, for your consideration.

Not finding the part at Digi-Key might be a separate issue. I suspect that my LibreOffice Calc installation may be having trouble with the .xlsx file created by recent KiCost versions. I'll continue working on this tomorrow, and I'll open a separate issue if needed. Should I expect LibreOffice Calc to be able to digest KiCost's .xlsx output, or is KiCost using features that rely on real Microsoft Excel?

mdeweerd, 2017-11-30:
That's not a good idea of those manufacturers - BOMs are used a lot as CSV files and this means that the fields need to be quoted.  That's probably generating some similar issues here and there!

NF6X, 2017-11-30:
I agree that it's not a good idea for manufacturers to use commas in their part numbers, and that KiCost probably isn't the only tool that has ever had a problem with the commas. But if major manufacturers like Nexperia (formerly NXP) and Toshiba are doing that, then we don't have much choice but to accommodate their behavior somehow. Hirose likes to use parentheses in their part numbers, and dashes are common. I don't think I've seen a semicolon in a part number yet, and I will be very annoyed if I ever do.

An alternative to disallowing commas as subpart separators might be to provide a way to escape special characters in the manf# field. I don't know if that would be a better or worse solution.

xesscorp, 2017-11-30:
Place a backslash before the comma or semicolon and it will be ignored as a multi-part component.

closed 2017-11-30 by xesscorp