hildogjr/KiCost #110: Include variant-specific user-requested fields in BOM

dogfinder, 2017-12-01:
I'm using the old-style variants feature for assembly variants which not only have some components selectively omitted, but also assign different part numbers to some parts based on the selected variant (e.g., for changing the tuning of analog filters in different variants). Not only am I changing the manf and manf# fields, but I'm also assigning different values to custom fields such as an internal part number. These custom fields are requested for inclusion in the BOM with the --fields command-line flag.

For the parts whose custom fields change based on assembly variant, the correct custom field values are not being assigned in the BOM. I believe I've fixed this by preventing the user-requested fields from being renamed with a "local:" prefix, much like the kicost-specific fields (manf, manf#, etc.). I'll issue a pull request with my proposed change.

One minor remaining nit is that since I don't see a way to have variant-specific values for the Value field, part numbers which are used both for common components and variant-specific components do not get grouped together in the BOM spreadsheet. Thus, these parts show up as multiple BOM line items. I don't have a suggestion for how to fix this yet.

While I'm here, PLEASE don't deprecate the old-style variants in favor of the new variant field! If I understand what I've read in the KiCad documentation, the new variant field only supports variant-specific inclusion/omission of parts, but not variant-specific changes of manf/manf#/etc. field values. I have a strong need to be able to change part numbers in assembly variants of the same PCB.

P.S.: I'm the same guy who has also contributed here as NF6X. I've made a new account for participating on GitHub in activities related to my day job.

hildogjr, 2018-02-24:
Dear Mark (@dogfinder ), could you update or pull-request or explain your changes on the KiCost code to I try to supply your request?

dogfinder, 2018-02-28:
The master branch has changed enough that I'll close this feature request rather than trying to merge my changes into it and updating my pull request. KiCost is really, really great when it works, but I've had many issues with the web scraping breaking at critical times. I've stopped using KiCost, and instead wrote a much simpler BOM formatter which fits my needs better.

closed 2018-02-28 by dogfinder

hildogjr, 2018-02-28:
Unfortunately, I agree, we been have a lot of issue (mainly) with the scrape modules. I still can not help yet, scrape is the part of KiCost that I understand less (maybe in the future, after some time of more study I will help).

dogfinder, 2018-02-28:
I hope that the scraping stuff will get solid again, because KiCost is great for my home projects where I'm the purchasing department. At work, I was mostly just using KiCost for BOM pretty-printing and rough price estimates, but I don't do any of the actual ordering.

hildogjr, 2018-03-01:
Think on this. A `--no_scrape` option would be nice. I know that exist better tools to create a pretty printing (to assemble list), but include this appear a nice option (when I don't want prices).

dogfinder, 2018-03-01:
Yes, I think that a --no-scrape option would be helpful.

The BOM generator program that I wrote for use instead of KiCost at the day job tries to solve a slightly different problem than KiCost does. It's built around my need to have each component in the BOM keyed, grouped, and sorted based on an internal company part number. It performs some consistency checks (e.g. inconsistent combinations of internal part number vs. manufacturer part number), generates separate BOMs for each assembly variant, and reports parts with "TBD" (to be determined) as the internal part number to help me keep track of which new parts need part numbers assigned. I don't know if it makes sense to have any such features in KiCost, but I mention it because it's a progression from the workflow that led me to make this feature request in the first place. It presently parses the same part fields that KiCost does, including the old-style variant field names, simply because I already put so much effort into adding all of those fields into the design that I wrote it for. Maybe I'll use a different field naming scheme in the future when I turn it into a more general-purpose tool, but for now it's a direct replacement for KiCost in the particular project I'm trying to push out the door.