hildogjr/KiCost #112: Not getting some parts

hildogjr, 2017-12-01:
Sometimes some existent components parts not appear in my spreadsheet, the part exist in the distributors but the scrap web module appear return `Empty/None`. I don't know if is related with #106.

For example, check for the item `MC14504BDTR2G` in my spreadsheet, it exist on Digikey but is empty in the file.

This error appear be intermittent and is not correlated with the part (the error shift between the parts).

I used `kicost -w -q -s --retires 100 file.xml` call.
[acquire-PWM.xlsx](https://github.com/xesscorp/KiCost/files/1521577/acquire-PWM.xlsx)
[acquire-PWM.xml.zip](https://github.com/xesscorp/KiCost/files/1521580/acquire-PWM.xml.zip)

dogfinder, 2017-12-01:
I am experiencing the same issue right now.

hildogjr, 2017-12-02:
The tests that I performed is related with the scrap web modules (if you use the debug option or add a `print()`) and, in my case, is increased using more parallels process (when I opened the issue I used the serialization `-s`option).

xesscorp, 2017-12-03:
I've seen this behavior many times before. Sometimes I'll scrape Digi or Mouser and get everything; the next scrape has missing pieces or, sometimes, nothing at all. It may be dependent upon how often I scrape a site within a given time period (such as during testing). I'm sure the distributors don't like being scraped. It probably violates their terms of service. So they might be doing some throttling if they see a lot of accesses to a particular part from the same IP.

In any event, I'm not sure this is a problem inside KiCost. I tried the acquire-PWM.xml file and the pricing for the MC14504BDTR2G came through fine from Digikey.

hildogjr, 2017-12-03:
Yep, I also think that's related with the distributor / Internet connection.
But maybe, improving #97 and adding a `--time-between-scrape` (in milliseconds default 0, a waiting time between sequential scraps), could we work around this "issue"?

xesscorp, 2017-12-05:
Some testing on Digikey showed that placing a 5-10 second delay between part scrapes raised the percentage of successful scrapes from 16% to >90% on a list of 400 parts.

hildogjr, 2017-12-05:
Interesting. I know that more time is annoying, but keep KiCost responsible.
We could add a heuristic, the standard retries is 4 so, something like increasing the time waiting after consecutives empty results. But will interesting to differentiate a `None` from a none existent part and from the "error" (I don't know if is possible).

hildogjr, 2017-12-17:
I am closing this due KiCost not presenting this issue anymore (in version `0.1.42`).

closed 2017-12-17 by hildogjr