hildogjr/KiCost #114: Farnell scrap module freeze process

hildogjr, 2017-12-07:
Every time that I scrap for prices using the farnell module the process still freeze in 0%, not going on or give me some error. I don't now if is a error in the scrap module or it just need to be updated to some web page change.
To work around this issue I am using `--include digikey mouser rs tme newark` or `--exclude farnell`.

adamheinrich, 2017-12-07:
I have the same problem with Farnell right now.

adamheinrich, 2017-12-07:
Quick observations:

Setting User-agent to a newer browser and adding the following headers helps:

```
    req.add_header('Accept', 'text/html')
    req.add_header('Cookie', 'foo=bar')
```

adamheinrich, 2017-12-07:
However, I'm not able to find a good database of User-agent strings to update the current one. THe one I successfully used to access Farnell using KiCost was `Chrome/62.0.3202.94`, the newest one at http://www.useragentstring.com/pages/useragentstring.php?name=Chrome is `Chrome/41.0.2228.0`.

adamheinrich, 2017-12-07:
This one looks good: https://techblog.willshouse.com/2012/01/03/most-common-user-agents/

PR in a few minutes...

closed 2017-12-08 by hildogjr