hildogjr/KiCost #129: Parts with Subparts multipliers crash

romain145, 2018-01-04:
Using multipliers for subparts cause the following issue.

Terminal error:
`Get schematic XML...
Get parts library...
Get components...
Search for subparts within parts...
Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 9, in <module>
    load_entry_point('kicost==0.1.42', 'console_scripts', 'kicost')()
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/__main__.py", line 198, in main
    scrape_retries=args.retries)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/kicost.py", line 135, in kicost
    p, info = eda_tool_module.get_part_groups(in_file[i_prj], ignore_fields, variant[i_prj])
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/eda_tools/kicad/kicad.py", line 197, in get_part_groups
    return group_parts(accepted_components), prj_info
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/eda_tools/eda_tools.py", line 110, in group_parts
    components = subpart_split(components)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/eda_tools/eda_tools.py", line 319, in subpart_split
    subpart_qty, subpart_part = subpart_qtypart(p_manf_code)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/eda_tools/eda_tools.py", line 424, in subpart_qtypart
    if len(re.sub('[\.\/]','',strings[0])) < re.sub('[\.\/]','',len(strings[1])):
  File "/usr/lib/python2.7/re.py", line 155, in sub
    return _compile(pattern, flags).sub(repl, string, count)
TypeError: expected string or buffer`

For example, I'm using a connector socket that needs a housing and crimps:
manf#: 649004121732; 649004013322; 64900613722:4
manufacturer: Wurth Electronics

If I remove the 3rd subpart, kicost runs fine.

hildogjr, 2018-01-05:
Could you attach your XML?
I tested with a part with just this specifications and worked fine.

romain145, 2018-01-05:
I just tried and confirm that this is now working with the git head https://github.com/xesscorp/KiCost/commit/8b2353a726bba462e48007071bd0e475ce380868

closed 2018-01-05 by romain145