hildogjr/KiCost #130: Install kicost without GUI

romain145, 2018-01-05:
Would it be possible to add an option to install kicost without the gui dependency?
I'm using the Windows 10 Linux Subsystem (headless Ubuntu) and would like to do without the wx_py and gtk+ dependencies.

hildogjr, 2018-01-05:
@romain145 , Are you install by a zip, pip or easy_installer?
Just to track, the GUI as a new addition to KiCost and may be some issues/dependences to fix.

romain145, 2018-01-05:
I used easy_installer.
I managed to remove all references to kicad_gui in the source and wx_py in the requirements in setup.py.
Just saying that it would be nice to have a CLI (no-gui) build target.

hildogjr, 2018-01-05:
Maybe the default installation just CLI, and adding the dependences (as a additional command on installation to the GUI).

hildogjr, 2018-01-08:
@romain145 . I fixed the dependency on installation of "KiCost GUI" <https://pypi.python.org/pypi/wxPython/4.0.0b2>.
Check if you have some issue yet.
The KiCost will always have the CLI default interface. The GUI just aims to reach users unfamiliar with the terminal. And by the `--user` parameter that you not need always to type the configurations in the call (using the already saved in the GUI).

romain145, 2018-01-08:
@hildogjr I understand what you mean and it makes sense to install it by default with the GUI dependency. However, for non-beginner users, I would much rather prefer to have the option to not include GUI for the following reason:

```
romain@Aquaputer:~$ sudo apt-get install libgtk-3-dev
[sudo] password for romain:
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  adwaita-icon-theme at-spi2-core autotools-dev colord colord-data dconf-gsettings-backend dconf-service debhelper
  dh-strip-nondeterminism fontconfig fontconfig-config fonts-dejavu-core gettext gir1.2-atk-1.0 gir1.2-atspi-2.0
  gir1.2-freedesktop gir1.2-gdkpixbuf-2.0 gir1.2-gtk-3.0 gir1.2-pango-1.0 glib-networking glib-networking-common
  glib-networking-services gsettings-desktop-schemas hicolor-icon-theme humanity-icon-theme intltool-debian
  libarchive-zip-perl libasprintf-dev libatk-bridge2.0-0 libatk-bridge2.0-dev libatk1.0-0 libatk1.0-data libatk1.0-dev
  libatspi2.0-0 libatspi2.0-dev libavahi-client3 libavahi-common-data libavahi-common3 libboost-filesystem1.58.0
  libboost-system1.58.0 libcairo-gobject2 libcairo-script-interpreter2 libcairo2 libcairo2-dev libcapnp-0.5.3
  libcolord2 libcolorhug2 libcroco3 libcups2 libdatrie1 libdbus-1-dev libdconf1 libdrm-amdgpu1 libdrm-common
  libdrm-dev libdrm-intel1 libdrm-nouveau2 libdrm-radeon1 libdrm2 libegl1-mesa libegl1-mesa-dev libepoxy-dev libepoxy0
  libexif12 libfile-stripnondeterminism-perl libfontconfig1 libfontconfig1-dev libfreetype6 libfreetype6-dev libgbm1
  libgd3 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common libgdk-pixbuf2.0-dev libgettextpo-dev libgettextpo0
  libgl1-mesa-dri libglapi-mesa libglib2.0-bin libglib2.0-dev libgmp-dev libgmpxx4ldbl libgphoto2-6 libgphoto2-l10n
  libgphoto2-port12 libgraphite2-3 libgtk-3-0 libgtk-3-bin libgtk-3-common libgudev-1.0-0 libgusb2 libharfbuzz-dev
  libharfbuzz-gobject0 libharfbuzz-icu0 libharfbuzz0b libice-dev libice6 libieee1284-3 libjbig0 libjpeg-turbo8
  libjpeg8 libjson-glib-1.0-0 libjson-glib-1.0-common liblcms2-2 libllvm5.0 libltdl7 libmail-sendmail-perl
  libmirclient-dev libmirclient9 libmircommon-dev libmircommon7 libmircookie-dev libmircookie2 libmircore-dev
  libmircore1 libmirprotobuf3 libpango-1.0-0 libpango1.0-dev libpangocairo-1.0-0 libpangoft2-1.0-0 libpangoxft-1.0-0
  libpciaccess0 libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpixman-1-0 libpixman-1-dev libpng12-dev
  libprotobuf-dev libprotobuf-lite9v5 libprotobuf9v5 libproxy1v5 libpthread-stubs0-dev librest-0.7-0 librsvg2-2
  librsvg2-common libsane libsane-common libsensors4 libsm-dev libsm6 libsoup-gnome2.4-1 libsoup2.4-1
  libsys-hostname-long-perl libthai-data libthai0 libtiff5 libtimedate-perl libtxc-dxtn-s2tc0 libunistring0 libvpx3
  libwayland-bin libwayland-client0 libwayland-cursor0 libwayland-dev libwayland-egl1-mesa libwayland-server0
  libx11-dev libx11-doc libx11-xcb-dev libx11-xcb1 libxau-dev libxcb-dri2-0 libxcb-dri2-0-dev libxcb-dri3-0
  libxcb-dri3-dev libxcb-glx0 libxcb-glx0-dev libxcb-present-dev libxcb-present0 libxcb-randr0 libxcb-randr0-dev
  libxcb-render0 libxcb-render0-dev libxcb-shape0 libxcb-shape0-dev libxcb-shm0 libxcb-shm0-dev libxcb-sync-dev
  libxcb-sync1 libxcb-xfixes0 libxcb-xfixes0-dev libxcb1-dev libxcomposite-dev libxcomposite1 libxcursor-dev
  libxcursor1 libxdamage-dev libxdamage1 libxdmcp-dev libxext-dev libxfixes-dev libxfixes3 libxft-dev libxft2
  libxi-dev libxi6 libxinerama-dev libxinerama1 libxkbcommon-dev libxkbcommon0 libxpm4 libxrandr-dev libxrandr2
  libxrender-dev libxrender1 libxshmfence-dev libxshmfence1 libxtst-dev libxtst6 libxxf86vm-dev libxxf86vm1 nettle-dev
  pkg-config po-debconf ubuntu-mono x11-common x11proto-composite-dev x11proto-core-dev x11proto-damage-dev
  x11proto-dri2-dev x11proto-fixes-dev x11proto-gl-dev x11proto-input-dev x11proto-kb-dev x11proto-randr-dev
  x11proto-record-dev x11proto-render-dev x11proto-xext-dev x11proto-xf86vidmode-dev x11proto-xinerama-dev
  xorg-sgml-doctools xtrans-dev
Suggested packages:
  colord-sensor-argyll dh-make gettext-doc autopoint libcairo2-doc cups-common libgd-tools libglib2.0-doc gmp-doc
  libgmp10-doc libmpfr-dev gphoto2 gvfs libgtk-3-doc libice-doc liblcms2-utils imagemagick libpango1.0-doc
  librsvg2-bin avahi-daemon hplip libsane-extras sane-utils lm-sensors libsm-doc libxcb-doc libxext-doc
  libmail-box-perl
The following NEW packages will be installed:
  adwaita-icon-theme at-spi2-core autotools-dev colord colord-data dconf-gsettings-backend dconf-service debhelper
  dh-strip-nondeterminism fontconfig fontconfig-config fonts-dejavu-core gettext gir1.2-atk-1.0 gir1.2-atspi-2.0
  gir1.2-freedesktop gir1.2-gdkpixbuf-2.0 gir1.2-gtk-3.0 gir1.2-pango-1.0 glib-networking glib-networking-common
  glib-networking-services gsettings-desktop-schemas hicolor-icon-theme humanity-icon-theme intltool-debian
  libarchive-zip-perl libasprintf-dev libatk-bridge2.0-0 libatk-bridge2.0-dev libatk1.0-0 libatk1.0-data libatk1.0-dev
  libatspi2.0-0 libatspi2.0-dev libavahi-client3 libavahi-common-data libavahi-common3 libboost-filesystem1.58.0
  libboost-system1.58.0 libcairo-gobject2 libcairo-script-interpreter2 libcairo2 libcairo2-dev libcapnp-0.5.3
  libcolord2 libcolorhug2 libcroco3 libcups2 libdatrie1 libdbus-1-dev libdconf1 libdrm-amdgpu1 libdrm-common
  libdrm-dev libdrm-intel1 libdrm-nouveau2 libdrm-radeon1 libegl1-mesa libegl1-mesa-dev libepoxy-dev libepoxy0
[... cut ...]
  x11proto-dri2-dev x11proto-fixes-dev x11proto-gl-dev x11proto-input-dev x11proto-kb-dev x11proto-randr-dev
  x11proto-record-dev x11proto-render-dev x11proto-xext-dev x11proto-xf86vidmode-dev x11proto-xinerama-dev
  xorg-sgml-doctools xtrans-dev
The following packages will be upgraded:
  libdrm2
1 upgraded, 234 newly installed, 0 to remove and 0 not upgraded.
Need to get 55.5 MB of archives.
After this operation, 375 MB of additional disk space will be used.
Do you want to continue? [Y/n] n
Abort.
romain@Aquaputer:~$
```

hildogjr, 2018-01-08:
Understood.
Maybe is better leave this disabled, for Linux, these are default libraries. But it are "heavy" packages space for Windows.
So, leaving disabled and look for some way to (not default) install the GUI dependences (using some additional command on KiCost installation).

closed 2018-01-09 by hildogjr