hildogjr/KiCost #131: Part Grouping issue

romain145, 2018-01-05:
The part grouping seems broken on the git head https://github.com/xesscorp/KiCost/commit/8b2353a726bba462e48007071bd0e475ce380868

As per the documentation, if only one of the parts has data, that data is propagated to all the other parts in the group.
See the following BOM extract for 100nF capacitor:
```
<comp ref="C1">
      <value>100n</value>
      <footprint>lib:C_0805</footprint>
      <fields>
        <field name="manf#">CC0805KRX7R9BB104</field>
        <field name="manf">Yageo</field>
      </fields>
      <libsource lib="device" part="C"/>
      <sheetpath names="/" tstamps="/"/>
      <tstamp>5671DB9D</tstamp>
    </comp>
<comp ref="C2">
      <value>100n</value>
      <footprint>lib:C_0805</footprint>
      <libsource lib="device" part="C"/>
      <sheetpath names="/" tstamps="/"/>
      <tstamp>56A37DE7</tstamp>
    </comp>
```

And the result in the Excel sheet:
```
C1 | 100n | lib:C_0805 | CC0805KRX7R9BB104 | 100 | $0.01
C2,C4,C16-C21,C46,C47,C64,C75 | 100n | lib:C_0805 |   | 1200 | $0.00
```

Am I missing something?

I attached the BOM and the Excel sheet.
[board.zip](https://github.com/xesscorp/KiCost/files/1606535/board.zip)

The command line is:
`kicost -i board --overwrite --debug 1 --exclude mouser newark`

hildogjr, 2018-01-05:
@xesscorp , maybe related with #102.

Other issue founded, the U14 part is
    <field name="manf#">ULN2803A; DS1009-18AT1NX</field>
    <field name="manf">ST, Connfly</field>

The "manf#" are split, but not the "manf". Maybe because the difference `;` and `,`?

hildogjr, 2018-01-13:
Founded the reason, because of the "manf#_qty" field generated by the feature of sub-parts.
I don't now how no user said before about this problem. It is generated also when you use field as "digikey#".
I make some restructuration in the grouping code, changing the first for and preparing the code to solve #102.

commit 050795c4 (2018-01-15): Fix #131 andon way of #102
closed 2018-01-13 by hildogjr