hildogjr/KiCost #139: STOP COMMITTING!

xesscorp, 2018-01-15:
hidogjr, please stop committing into this repo for the time being.

I've checked out the current HEAD and tried it on my Windows 7 PC. It throws errors as soon as I start it under Python 2.7.14. Under Python 3.6, it works correctly in single-processor mode, but throws an error when run in multiprocessor mode.

I have a version that runs correctly under 2.7 and 3.6, but I can't check it in until I resolve all the commits you've made since I started doing my modifications.

For now, do not commit any more changes. I'll let you know when it's safe to commit again.

hildogjr, 2018-01-15:
Ok

xesscorp, 2018-01-15:
OK, you can continue development.

closed 2018-01-15 by xesscorp