hildogjr/KiCost #144: Errors that occur with the kicost GUI should appear in a dialog window.

xesscorp, 2018-01-15:
Errors/warnings/log messages should appear in a dialog window when the kicost GUI is used. (Not in the terminal window.)

hildogjr, 2018-01-16:
Merging with #143, add also a process bar.
"Maybe, call kicost as a thread and deal with the output to generate the process bar and the messages/warning dialog."

hildogjr, 2018-02-24:
For the process bar update process, wait for `tqdm 5`? https://github.com/tqdm/tqdm/issues/172

hildogjr, 2018-03-10:
Necessary (better way) redirect the sdtout to a class that overwrite the `sys.stdout` (https://stackoverflow.com/questions/21341096/redirect-print-to-string-list) and use this on the `multiprocess.call(kicost())`. But this is creating some issues when run as parallel scraping.

hildogjr, 2018-03-20:
Change the print output to message box on the GUI, but missing fix related with #193 to work fine.

hildogjr, 2018-06-23:
For reference
https://www.thecodingforums.com/threads/wxpython-redirect-the-stdout-to-a-textctrl.515815/

commit 72a08649 (2018-01-17): Merge pull request #153 from hildogjr/master

Fixed #144 #145 and improved GUI
commit 3661f8d3 (2018-03-20): Merge pull request #207 from hildogjr/master

Fix #144, but missing process bar and issue #193
commit 9be77920 (2018-03-20): Fix #144, but missing process bar and issue #193
commit 62e820f1 (2018-06-30): Fix #144
closed 2018-06-30 by hildogjr